AdaCore/svd2ada #2: CONSTRAINT_ERROR : s-valllu.adb:291 explicit raise

TomaszOsypinski, 2016-01-29:
I have tried to generate Ada code for the Infineon XMC4500 microcontroller with svd2ada but I got an error.

H:\ravenscar-sfp\svd2ada>svd2ada XMC4500.svd -o XMC4500
*** WARNING: ignoring device element vendor
*** WARNING: ignoring device element cpu

raised CONSTRAINT_ERROR : s-valllu.adb:291 explicit raise

XMC4500.svd file is form http://www.keil.com/dd2/pack/

Maybe you will be interesting to solve this problem for svd2ada.

Best Regards,
T.O.

[XMC4500.zip](https://github.com/AdaCore/svd2ada/files/110051/XMC4500.zip)

Matthias-R, 2016-01-29:
Not a solution, but a comment and a question:
In this case, it's obvious that the SVD file from Infineon contains elements which are not yet implemented in svd2ada. These elements can be easily found in the first few lines of the file.

Others incompatibilities might be less obvious. Would it be possible to include the svd file line number when an error occurs? This question already came up during my experiments with my own, simple svd2ada converter. I don't know if this is possible at all with xmlada.

Regards,

Matthias

TomaszOsypinski, 2016-02-01:
Problem with XMC4500.svd file is in enumeratedValue tag. Value of this tag is done in binary system with notation like this #binaryValue for example #0101 = 0x5.

So in function Descriptors.Enumerate.Read_Value for the case Tag = "value" is problem with function Base_Types.Get_Value (Child) which does not service binary case.

Service this case will fix this problem.

Regards,
Tom

lambourg, 2016-02-01:
Hello Tom,

OK, I'm downloading this file and will fix in svd2ada.

Thanks for the report!

Best regards,
- Jerome

lambourg, 2016-02-01:
> Le 1 févr. 2016 à 09:39, Tomi <notifications@github.com> a écrit :
>
> Problem with XMC4500.svd file is in enumeratedValue tag. Value of this tag is done in binary system with notation like this [...]

I have indeed a local fix now to take into account #[0-9], but generation still
fails because of some incorrect values in XMC4500.svd:

The enumatated value is according to the CMSIS-SVD schema a
scaledNonNegativeInteger type defined as:

 <xs:simpleType name="scaledNonNegativeInteger">
    <xs:restriction base="xs:string">
      <xs:pattern value="[+]?(0x|0X|#)?[0-9a-fA-F]+[kmgtKMGT]?"/>
    </xs:restriction>
  </xs:simpleType>

However, we find in this svd file:

<enumeratedValue>
  <name>value1</name>
  <description>Disconnected</description>
  <value>#000X</value>
</enumeratedValue>

This is supposed (I guess) to express that any number should match,
but can't be expressed easily in Ada, and will not pass the schema verification
anyway (that I'm currently implementing).

TomaszOsypinski, 2016-02-01:
Jerome,

I think you should ignore case with character X at the end of the string. It is not compatible with CMSIS-SVD scheme. Maybe some warring should be occurred.

BTW it is only one place with value #000X in XMC4500.svd (DSRC bit fields).
In reference manual for XMC4500 (V1.5 2014-07) for this bit fields X means 0 or 1 :

DSRC [3:0] rw Input Data Source Select
000XB Disconnected
0010B External, from input A, direct
0011B External, from input A, inverted
0100B External, from input B, direct
0101B External, from input B, inverted
Other combinations are reserved

And in headr file  xmc_dsd.h form XMClib v2.1.2, it was used 0 for case #000X.
typedef enum XMC_DSD_CH_DATA_SOURCE
{
  XMC_DSD_CH_DATA_SOURCE_DISCONNECT  = 0U,  /**< Disconnected */
  XMC_DSD_CH_DATA_SOURCE_A_DIRECT    = 2U,  /**< External source, from input A, direct */
  XMC_DSD_CH_DATA_SOURCE_A_INVERTED  = 3U,  /**< External source, from input A, inverted */
  XMC_DSD_CH_DATA_SOURCE_B_DIRECT    = 4U,  /**< External source, from input B, direct */
  XMC_DSD_CH_DATA_SOURCE_B_INVERTED  = 5U   /**< External source, from input B, inverted */
} XMC_DSD_CH_DATA_SOURCE_t;

I changed it to #0000 in XMC4500.svd and put to svd2ada (with my fix of this issue) and svd2ada works ok.

Matthias-R, 2016-02-01:
Hello Jérôme,

the #000X seems to be legal.

I found in http://www.keil.com/pack/doc/CMSIS/SVD/html/group__svd___format__gr.html 
(CMSIS-SVD  Description, Version 1.3.1)
in section CMSIS-SVD/Reference/SVD File Schema Levels:

> **Constants**
> 
> Number constants shall be entered in hexadecimal, decimal, or binary format.
>
>    The Hexadecimal format is indicated by a leading "0x".
>    The Binary format is indicated by a leading "#".
>    All other formats are interpreted as decimal numbers.
>    The value tag in enumeratedValue accepts 'do not care' bits represented by "x".

I think "scaledNonNegativeInteger" for enumerated value is an error in the SVD documentation. According to the schema file, for the value of an <enumeratedValue> "enumeratedValueDataType" shall be used, not "scaledNonNegativeInteger"
And "enumeratedValueDataType" allows the "x"

It would be interesting what the original SVDConv.exe does in this case...

This register seems to be a configuration register which is probably only written by the program. In this case, the "x" can safely be replaced by a zero (or a one, you have the choice :-) ).

It gets problematic if both values can occur because the bit is written by the hardware
(and both values shall have the same meaning). I have no idea how to express that in Ada...

Regards,

Matthias

lambourg, 2016-02-01:
Ok, so I leave this case opened for now. I need to think a bit more about it.

Maybe I'll need to check first for the easy case (write-only registers) in which case I'll just fill the X with zeros. The read-only or read/write registers will need more tweaks, I guess I'll issue a warning first in this case before we find a nice solution.

lambourg, 2016-02-01:
BTW, to Tomasz, the binary case not containing 'X' is now properly handled, so you just need to get up-to-date sources to solve your issue (fixing the case with #000X in the svd file).

lambourg, 2016-02-01:
finally, to go back to your initial comment, Matthias, unfortunately I'm using the DOM parser of XmlAda, and this removes all relationship with the original file. To keep this information, I'd have to switch to the SAX parser which is a bit more complex to handle. This might be worth it if we can get meaningful errors though, so I'll keep that in mind and see if I can do something about it.

Matthias-R, 2016-02-02:
I looked in the documentation for XML/Ada in the meantime. Section 5.4 sounds to me as if it's possible to add line number information to the DOM tree. If that works, you don't need to switch to the SAX parser.

The 'X' is indeed no problem for write-only registers. Unfortunately, these are quite rare. But many read-write registers are in fact only written by the program - a read reads back the content written before. In this case, replacing X by zero would be OK, too. If the bits in question can change without actions from the program (= are changed by the hardware), then this wouldn't work. A warning is really necessary then.

lambourg, 2016-07-11:
This should now finally be properly handled by svd2ada. So I'm closing this issue. Thanks for your report!

closed 2016-07-11 by lambourg