AdaCore/svd2ada #3: Doesn’t detect use of keyword as default for field

simonjwright, 2016-02-02:
Hi,

I’ve been working on using svd2ada with Atmel devices (specifically, the ATSAM3X8E in the Arduino Due).

ATSAM3X8E.svd contains a field with enumerates `OUT`, `IN`. The _type_ is correctly generated (`Out_K`, `In_K`) but the default value in the register declaration remains `OUT`.

I don’t think it will be easy to fix this with the current design? at descriptors-field.adb:294,

                        if Val.Value = Default then
                           Default_Id := Val.Name;
                        end if;

                        Add_Enum_Id
                          (Enum_T,
                           Id      => To_String (Val.Name),
                           Repr    => Val.Value,
                           Comment => To_String (Val.Descr));

means that any adjustments made to `Val.Name` during `Add_Enum_Id` won’t be reflected in `Default_Id`.

The way I got round this in [my fork](https://github.com/simonjwright/svd2ada/tree/sjw) was to have a new public function `Base_Types.Get_Name` (which does much the same as your `Protect_Keywords`) and change all the places where `Get_Value` is called to retrieve a name to `Get_Name`. I also handled the case where the leading character of the enumerate is a digit - which has a similar problem - in `Descriptors.Field’, same place as above:

                        if Val.Value = Default then
                           if Unbounded.Element (Val.Name, 1) in '0' .. '9' then
                              Default_Id := Enum_Name & "_" & Val.Name;
                           else
                              Default_Id := Val.Name;
                           end if;
                        end if;

lambourg, 2016-02-02:
Hello Simon,

This should now be fixed. Please tell me if the latest commit indeed fixes the issue you are seeing !

Best regards,
- Jerome

closed 2016-02-02 by lambourg

simonjwright, 2016-02-02:
Hi Jerome,

Yes, that did it!

Thanks,

--Simon