AdaCore/svd2ada #4: Bad default value for enumerated field

simonjwright, 2016-02-03:
If an enumerated field has no enumerate whose value (representation) is 0, a default of `16#0#` is created.

Given this fragment 

            <field>
              <name>FKEY</name>
              <description>Flash Writing Protection Key</description>
              <bitOffset>24</bitOffset>
              <bitWidth>8</bitWidth>
              <access>write-only</access>
              <enumeratedValues>
                <enumeratedValue>
                  <name>PASSWD</name>
                  <description>The 0x5A value enables the command defined by the bits of the register. If the field is written with a different value, the write is not performed and no action is started.</description>
                  <value>0x5a</value>
                </enumeratedValue>
              </enumeratedValues>
            </field>

we get

```
   --  Flash Writing Protection Key
   type FKEY_Field is
     (Passwd --  The 0x5A value enables the command defined by the bits of the register. If the field is written with a different value, the write is not performed and no action is started.
     ) with Size => 8;
   for FKEY_Field use
     (Passwd => 90);

   --  EEFC Flash Command Register
   type FCR_Register is record
      --  unspecified
      Reserved_0_23 : Magic.UInt24 := 16#0#;
      --  Flash Writing Protection Key
      FKEY          : FKEY_Field := 16#0#;                     <<<<<<<<<<<<<<<<
   end record
     with Volatile, Size => 32, Bit_Order => System.Low_Order_First;

```

I think it would be better not to provide a default at all for such a field. Alternatively, make it the first enumerate?

lambourg, 2016-02-03:
Interesting case. The intent behind the 'default values' is to have an easy way to reset a register (e.g. those are always set to the register's reset value): Perif.Reg := (others => <>); resets the register without having to even read the MCU reference manual.
So my initial feeling is not to modify this behavior but instead do the following: if the reset value is not part of the enumerate field, this means that we should not map it to an Ada enum, but just make it a constant here:

`   subtype FKEY_Field is new Byte;
   FKEY_Passwd : constant FKEY_Field := 90;

   type FCR_Register is record
      --  unspecified
      Reserved_0_23 : Magic.UInt24 := 16#0#;
      --  Flash Writing Protection Key  FKEY : FKEY_Field := 16#0#
   end record;`

simonjwright, 2016-02-03:
I can see where you’re coming from, and (aside from mistakes in the SVD) this is the only place in ATSAM3X8E where there’s a _writable_ enumerate field without a zero value. There are a few other places where there’s a _read-only_ register with an enumerate field with no non-zero representation; e.g. in the CHIPID peripheral there is
```
   --  Embedded Processor
   type EPROC_Field is
     (Arm946Es, --  ARM946ES
      Arm7Tdmi, --  ARM7TDMI
      Cm3, --  Cortex-M3
      Arm920T, --  ARM920T
      Arm926Ejs, --  ARM926EJS
      Ca5, --  Cortex-A5
      Cm4 --  Cortex-M4
     ) with Size => 3;
   for EPROC_Field use
     (Arm946Es => 1,
      Arm7Tdmi => 2,
      Cm3 => 3,
      Arm920T => 4,
      Arm926Ejs => 5,
      Ca5 => 6,
      Cm4 => 7);
```
Would it be possible/reasonable to suppress initial values for read-only registers?

lambourg, 2016-02-03:
> Would it be possible/reasonable to suppress initial values for read-only registers?

This was on my todo list indeed.

simonjwright, 2016-02-03:
I said "a _writable_ enumerate field without a zero value” but of course what I meant was “without an enumerate corresponding to the default value”.

I suppose that in that case one could add such an enumerate called e.g. `Default_nnn` or `Reset_nnn`.

simonjwright, 2016-02-04:
Something like the attached (which only goes part way; should really insert the correct default value).

[src-descriptors-field.adb.txt](https://github.com/AdaCore/svd2ada/files/117050/src-descriptors-field.adb.txt)

lambourg, 2016-02-04:
The thing is this default value may be a non-zero value.

simonjwright, 2016-02-04:
This version of the earlier patch inserts a dummy default value (if required) in the correct place in the enumeration. The default value may be non-zero.

Patch: [src-descriptors-field.adb.txt](https://github.com/AdaCore/svd2ada/files/117408/src-descriptors-field.adb.txt)

Test: [magic.svd.txt](https://github.com/AdaCore/svd2ada/files/117419/magic.svd.txt)

lambourg, 2016-02-04:
Ok, I slightly changed your patch, and the issue should now be fixed. Thanks again for the report and for being an early tester for svd2ada :) I took the opportunity to also fix the incorrect generation of default field values for read-only fields.
I'm closing this issue then.

closed 2016-02-04 by lambourg