AdaCore/svd2ada #5: svd2ada does not understand ATSAMD21G18A.svd

tkoskine, 2016-03-11:
I am trying to generate SAMD21 packages from ATSAMD21G18A.svd file, but svd2ada (rev 3d8aed06aa74dab092a6c0fdac8b8c6fbdc8cbdf) gives me errors:
```
D:\Code\git\svd2ada\test
> ..\svd2ada -o out ATSAMD21G18A.svd
*** WARNING: ignoring device element vendor
*** WARNING: ignoring device element vendorID
*** WARNING: ignoring device element series
*** WARNING: ignoring device element licenseText
*** WARNING: ignoring device element cpu
*** WARNING: ignoring device element headerSystemFilename
*** WARNING: Unsupported register naming schema: COMPCTRL%s
*** WARNING: Unsupported register naming schema: SCALER%s
*** WARNING: Unsupported register naming schema: DCC%s
*** WARNING: Unsupported register naming schema: ENTRY%s
*** WARNING: Unsupported register naming schema: CONFIG%s
*** WARNING: Unsupported register naming schema: PRAS%s
*** WARNING: Unsupported register naming schema: PRBS%s
*** WARNING: Unsupported register naming schema: SFR%s
*** WARNING: Unsupported register naming schema: CLKCTRL%s
*** WARNING: Unsupported register naming schema: DATA%s
*** WARNING: Unsupported register naming schema: SERCTRL%s
*** WARNING: Unsupported register naming schema: CTRL%s
*** WARNING: Unsupported register naming schema: DIR%s
*** WARNING: Unsupported register naming schema: DIRCLR%s
*** WARNING: Unsupported register naming schema: DIRSET%s
*** WARNING: Unsupported register naming schema: DIRTGL%s
*** WARNING: Unsupported register naming schema: IN%s
*** WARNING: Unsupported register naming schema: OUT%s
*** WARNING: Unsupported register naming schema: OUTCLR%s
*** WARNING: Unsupported register naming schema: OUTSET%s
*** WARNING: Unsupported register naming schema: OUTTGL%s
*** WARNING: Unsupported register naming schema: PINCFG0_%s

raised CONSTRAINT_ERROR : register 'derivedFrom' is not known: PINCFG0_%s
D:\Code\git\svd2ada\test
>

```

The problematic part is something like this:
```
<register derivedFrom="PINCFG0_%s">
          <dim>32</dim>
          <dimIncrement>0x1</dimIncrement>
          <name>PINCFG1_%s</name>
          <description>Pin Configuration n - Group 1</description>
          <addressOffset>0xc0</addressOffset>
        </register>
```

Example .svd file is attached.

[ATSAMD21G18A-svd.txt](https://github.com/AdaCore/svd2ada/files/169771/ATSAMD21G18A-svd.txt)

simonjwright, 2016-03-12:
SVD2Ada thinks that the only way of specifying an array of register components (e.g. array of bits) is using `name[%s]` syntax. But as far as I can see from the schema ([1.0](http://www.disca.upv.es/aperles/arm_cortex_m3/curset/CMSIS/Documentation/SVD/html/group__schema__gr.html), [1.1draft](http://www.disca.upv.es/aperles/arm_cortex_m3/curset/CMSIS/Documentation/SVD/html/group__schema__1__1__gr.html) - both somewhat unofficial) `name%s` is equally valid. Whether it means the same thing is a different matter.

lambourg, 2016-03-12:
Indeed, this particular SVD file is a good torture example for svd2ada, for at least two reasons:
* the array of registers is indeed different, but easy to fix
* it makes use of clusters of registers, which is not implemented yet.

This second issue will take a bit more time to fix unfortunately, as I'll need to reorder a bit the internal structures. Moreover, since SVD 1.3, it seems that we can now have clusters of clusters.

lambourg, 2016-08-09:
I finally had time to work on the 'cluster' support in svd2ada. This should work fine now, but of course if you find bugs, feel free to report here!

I'm closing this issue then.
Best regards,
Jerome

closed 2016-08-09 by lambourg