AdaCore/svd2ada #7: Style error on field

simonjwright, 2016-03-12:
If a field has no description, e.g.

            <field>
              <name>TSTP</name>
              <bitOffset>23</bitOffset>
              <bitWidth>1</bitWidth>
              <access>read-only</access>
            </field>

then the generated code has a comment with a trailing space, which gives a style error with -gnatpg.

    476.       --  Read-only. 
                             |
         >>> (style) trailing spaces not permitted
 
    477.       TSTP          : SR_TSTP_Field;

lambourg, 2016-03-12:
Hi Simon,

Thank you very much for your sanity checks ! This issue should now be fixed. Can you check?

Best regards,
- Jerome

closed 2016-03-12 by lambourg

simonjwright, 2016-03-12:
Jerome,

Thanks for quick fix! Minor problem (which results in CE!), see patch attached.

[strip_string.diff.txt](https://github.com/AdaCore/svd2ada/files/170646/strip_string.diff.txt)

simonjwright, 2016-03-13:
Now fixed, thanks.

lambourg, 2016-03-13:
You're welcome!