AdaCore/svd2ada #8: Enumerals WORD, BYTE conflict with base types

simonjwright, 2016-03-12:
ATSAM3X8E.svd contains, for example,

            <field>
              <name>WORD</name>
              <description>Word Transfer</description>
              <bitOffset>4</bitOffset>
              <bitWidth>1</bitWidth>
              <access>read-write</access>
              <enumeratedValues>
                <enumeratedValue>
                  <name>HALF</name>
                  <description>Half-Word transfer</description>
                  <value>0</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>WORD</name>
                  <description>Word Transfer</description>
                  <value>1</value>
                </enumeratedValue>
              </enumeratedValues>
            </field>

which leads to

     55.    --  Word Transfer
     56.    type WORD_Field is
     57.      (
     58.       --  Half-Word transfer
     59.       Half,
     60.       --  Word Transfer
     61.       Word)
     62.      with Size => 1;
     63.    for WORD_Field use
     64.      (Half => 0,
     65.       Word => 1);

and later to

    777.       --  Transmit Pointer Register
    778.       TPR  : Word;
                      |
         >>> subtype mark required in this context
         >>> found "Word" declared at line 61

This used not to be a problem, because `Base_Types.Target_Type` - as called - generated the fully-qualified name.

Same result with & without `--old`.

lambourg, 2016-03-13:
Hi Simon,

This should now be fixed: the base types are now always fully qualified.

simonjwright, 2016-03-13:
I think they maybe shouldn’t be _always_ fully qualified :-(

In the top level atsam3x8e.ads,

     12. package ATSAM3X8E is
     13.    pragma Preelaborate;
     14.
     15.    ---------------
     16.    -- Base type --
     17.    ---------------
     18.
     19.    subtype ATSAM3X8E.Word is Interfaces.Unsigned_32;
                             |
         >>> missing "is"

     20.    subtype ATSAM3X8E.Short is Interfaces.Unsigned_16;
                             |
         >>> missing "is"

etc (GNAT GPL 2015, FSF GCC 5.2.0 - not perhaps the clearest of error messages!)

lambourg, 2016-03-13:
Arf ! (headbang), ok, fixing this asap.

lambourg, 2016-03-13:
This should now be really fixed.

closed 2016-03-13 by lambourg

simonjwright, 2016-03-13:
Good, all OK now. Thanks!