AdaCore/svd2ada #9: Overlay of register array with single register

simonjwright, 2016-03-13:
In ATSAM3X8E.svd, some register arrays in UOTGHS (for example, `DEVEPTISR[%s]`; but not `DEVEPTCFG[%s]`) are overlaid by a single register definition (for example, `DEVEPTISR0_ISOENPT`). (I think this is an error on Atmel’s part).

The result is that we end up declaring

    type DEVEPTISR_Discriminant is
      (Default,
       Isoenpt);

    type DEVEPTISR_Aliased_Register
      (Disc : DEVEPTISR_Discriminant := Default)
    is record
       case Disc is
          when Default =>
             Default : DEVEPTISR_Registers;
          when Isoenpt =>
             ISOENPT : DEVEPTISR0_ISOENPT_Register;
       end case;
    end record
      with Unchecked_Union;

    for DEVEPTISR_Aliased_Register use record
       Default at 0 range 0 .. 31;
       ISOENPT at 0 range 0 .. 31;
    end record;

which is an error because `Default` has size 320 bits, not 32.

We could of course resolve this by patching the SVD to remove the offending single-register definitions.

lambourg, 2016-03-13:
Have you checked the reference manual/datasheet for the MCU? What does it say for those registers?

If the issue is only within the SVD (e.g. non compliance with the datasheet), then you should patch the SVD files. This already happened to me on some of the SVD files I've used (with some registers incorrectly placed). There's not much svd2ada can do in such case.

simonjwright, 2016-03-13:
There’s no *ISOENPT in the datasheet at all. Patching-out it is!

closed 2016-03-13 by simonjwright

lambourg, 2016-03-13:
Ok, fine! It's not the first time SVD files contain errors. On the contrary, most of them are erroneous at some point... That's why I started to commit "fixed" SVD files in the CMSIS-SVD folder. I've putted the fixed under quotes, because my testing is far from covering 100% of the files so far, so they might still contain errors. If you want your svd file to be tracked here, just do a pull request, I'll integrate it to the repository.

simonjwright, 2016-04-09:
I don’t know whether it’s possible to pull specific commits from another repository? if so,

Original SVD from ARM
https://github.com/simonjwright/svd2ada/commit/747ce3138022ad6a824b477ad2eceec69393f634

Modification to remove problems
https://github.com/simonjwright/svd2ada/commit/aadcc69812b8722a0aa7160081303c19ffb1b275