AdaCore/svd2ada #10: Array index range may not start from 0

simonjwright, 2016-03-26:
For ATSAM3X8E, the peripheral clock enable register 0 has bits 0 and 1 marked as “-“ and starts with Peripheral ID 2 in bit 2.

The datasheet:

[image]

The SVD says

        <register>
          <name>PMC_PCER0</name>
          <description>Peripheral Clock Enable Register 0</description>
          <addressOffset>0x00000010</addressOffset>
          <size>32</size>
          <access>write-only</access>
          <fields>
            <field>
              <name>PID2</name>
              <description>Peripheral Clock 2 Enable</description>
              <bitOffset>2</bitOffset>
              <bitWidth>1</bitWidth>
              <access>write-only</access>
            </field>

but the Ada has

    --  PMC_PCER0_PID array
    type PMC_PCER0_PID_Field_Array is array (0 .. 29) of PMC_PCER0_PID_Element
      with Component_Size => 1, Size => 30;

I think it would be much better if the array was (2 .. 31) - i.e. start the range at the numeric part of the name of the first element, here 2 from PID2.

lambourg, 2016-04-07:
Hi Simon,

This should now be fixed. From a similar SVD file (ATSAMG55J19.svd), svd2ada now generates the following:

```
   --  PMC_PCER0_PID array
   type PMC_PCER0_PID_Field_Array is array (8 .. 29) of Boolean
     with Component_Size => 1, Size => 22;
```

closed 2016-04-07 by lambourg

simonjwright, 2016-04-09:
Looking good here! thanks.