AdaCore/svd2ada #11: Undesirable effect of using “prependToName"

simonjwright, 2016-04-09:
In ATSAM .svd files, Atmel use the prependTo_name tag freely, and svd2ada now takes account of this.

There are 2 unfortunate results: looking at PWM, where we have

    <peripheral>
      <name>PWM</name>
      <version>6343I</version>
      <description>Pulse Width Modulation Controller</description>
      <prependToName>PWM_</prependToName>
 
we end up with

    type PWM_CLK_Register is record
    ...

and 

    type PWM_Peripheral is record
       --  PWM Clock Register
       CLK      : PWM_CLK_Register;

where the PWM_ in PWM_CLK_Register is overkill (at least in Ada).

More annoyingly, in USART, the prependToName tag is USART1_ for USART1, USART2 for USART2 etc, so that we end up with (for ATSAMG55J19)

    type USART_Peripheral
      (Mode : FlexCom_Mode := Usart_Mode)
    is record
       --  USART Receive Holding Register
       RHR          : USART5_RHR_Register;                   <<<<<<<<<<<
       --  USART Transmit Holding Register
       THR          : USART5_THR_Register;                    <<<<<<<<<<<

lambourg, 2016-04-11:
Even in Ada, this is not overkill in the case where different peripherals are defined in the same package (due to the <groupName> element), and where register types of those peripherals have name clashes.

In ATSAMG55J19 for example, the SYSC package defines multiple 'MR' registers: one for the RTT peripheral, one for the WDT peripheral, one for the RTC peripheral. If there's no prefix, then there's a type definition clash.

I agree that the side effect on the USART peripheral is unfortunate, and I'm trying currently to figure out a solution for that, but it's still better than compilation errors due to type clashes.

lambourg, 2016-04-11:
That being said, if you find a better way to fix the SYSC package (I was thinking of introducing nested packages eventually, but it's still not ideal), I'm ready to accept your contributions of course :)

closed 2016-04-24 by lambourg