AdaCore/svd2ada #12: Problem with aliased registers

Matthias-R, 2016-04-10:
LPC800 from NXP has some pairs of registers (a read-only and a write-only register) at the same address.
For example, SUM and WR_DATA are the same address. SUM is a read-only register, WR_DATA a write only register. Both registers are containing a single 32 bit field.

svd2ada cannot handle aliased registers if their names are too dissimilar. I got this error message when I try to process LPC800_v0.3.svd:

> *** WARNING ***: cannot find a proper Ada type name for the aliased registers 
> SUM and WR_DATA
> 
> raised CONSTRAINT_ERROR : descriptors-register.adb:252 index check failed

Regards,

Matthias

lambourg, 2016-04-11:
Ok, I knew it would appear at some point :(

I introduced recently 'helpers' files to workaround this kind of limitation. For now it allows applying a discriminent at the peripheral level (such as Host/Device mode for USB peripherals), and thus have the registers properly sorted there (no need for a discriminent and a common name at the register level). So when applicable, svd2ada will not try to figure out prefixes. This mechanism however only supports one discriminent for the moment (an enum). You can have a look at examples in the repository within the CMSIS-SVD folder, as svd2ada is still missing a proper documentation.

Another workaround would be to "patch" the SVD file to either add a prefix, or have a single definition such as "SUM_or_WR_DATA" for example?

Matthias-R, 2016-04-12:
I am not sure what will be the 'best' way. I don't really like the additional layer in the name hierarchy (added by the variant record) which has no correspondent in the data sheet of the processor. But I have no idea how to do it better...

Matthias-R, 2016-04-18:
I made some experiments with unchecked unions to find out what's possible and what's not (and yes - I also read the fine manual...)
What do you think about this version?
```ada
type Access_Mode is (RO, WO);

--  Cyclic Redundancy Check (CRC) engine
type CRC_Peripheral (A : Access_Mode := RO) is record
   --  CRC mode register
   MODE    : MODE_Register;
   --  CRC seed register
   SEED    : LPC800.Word;
   case A is
     when RO =>
       --  CRC checksum register
       SUM     : LPC800.Word;
     when WO =>
       --  CRC data register
       WR_DATA : LPC800.Word;
   end case;
end record
  with Volatile, Unchecked_Union;

for CRC_Peripheral use record
   MODE    at 0 range 0 .. 31;
   SEED    at 4 range 0 .. 31;
   SUM     at 8 range 0 .. 31;
   WR_DATA at 8 range 0 .. 31;
end record;

--  Cyclic Redundancy Check (CRC) engine
CRC_Periph : aliased CRC_Peripheral
  with Import, Address => CRC_Base;
```

In this example, the aliased register is the last one in the record anyway. Otherwise, the order of the registers must be rearranged since Ada requires the variant part of a record to be at its end. The correct physical order would then be defined in the representation clause.

With this declaration, the aliased registers can be accessed exactly the same way as the 'normal' registers, without an additional layer in the name hierarchy:
```ada
CRC_Periph.SEED    := 16#DEADBEEF#;
CRC_Periph.WR_DATA := 16#01234567#;
```

lambourg, 2016-04-18:
Ok, looks good. What you need then to produce such register is to create a LPC800_v0.3.svd2ada file with the following content:
`<?xml version="1.0"?>
<svd2ada apply="LPC800_v0.3.svd">
  <peripheral name="CRC">
    <discriminent id="A" type="Access_Mode">
      <value>RO</value>
      <value>RW</value>
    </discriminent>
    <register name="SUM" when="RO"/>
    <register name="WR_DATA" when="RW"/>
  </peripheral>
</svd2ada>`

The file needs to be place at the same location than your svd file. Tell me if this works for you !

Matthias-R, 2016-04-19:
Great!
That generates exactly the same declarations as my hand-written version (not counting some formatting differences).
I'll check if other aliased registers in this file follow the same pattern (two registers with the same address, one read-only, the other write-only). In fact, if you don't have some kind of mode switching, this is the only sensible case for two registers sharing the same address. Perhaps it would be possible to detect this pattern automatically without the *.svd2ada helper file?!?
The access mode of a register (read-write, read-only, write-only) is given (should be...) in the SVD file.

On minor point, not really important: The casing of 'RO' and 'WO' is not taken from the helper file, it is changed to 'Ro' and 'Wo'. That looks somewhat strange...

lambourg, 2016-04-19:
That is far from the only case where two registers overlap. Typical situation is : USB peripheral in host or device mode, Communication peripheral in SPI/I2C/USART mode (such as happen with some SAM MCU), etc.

Those overlapping registers are reported by svd2ada when generating the bindings. In general, those overlapping registers have some common prefix however that allows us to find proper names automatically.

Now on the casing issue, the enum values are automatically put in camel case, regardless they come from the SVD file or from the helper file. In case of helper file, the casing should be kept. Nice catch !

Matthias-R, 2016-04-20:
> That is far from the only case where two registers overlap. Typical situation is : USB peripheral in host or device mode, Communication peripheral in SPI/I2C/USART mode (such as happen with some SAM MCU), etc.

I see. If peripherals have multiple modes, there are infinite possibilities.
These peripherals have multiple registers with the same address, but only one at a time is used. The active register is selected with some kind of mode switching mechanism (e.g. a bit in another register).

My example is a bit different: There is no mode switching. Two registers share the same address, and both registers are used at the same time. In this case my statement remains valid: This type of register overlay makes only sense if one register is read-only and the other write-only.
This arrangement may occur in every 'normal' (single-mode) peripheral. For example, it's quite common to have RX data and TX data registers for serial ports (UART, SPI, I²C, etc...) at the same address. 
This has not very high priority (the manual intervention via the *.svd2ada file is a viable workaround) but nevertheless it would be nice to handle this case automatically.

Matthias-R, 2016-04-20:
I made some further investigations on the aliased registers in the LPC800 example.

> This has not very high priority (the manual intervention via the *.svd2ada file is a viable workaround) but nevertheless it would be nice to handle this case automatically.

Priority is low, indeed. The LPC800 has exactly one register pair of this type.
The two other occurrences of overlapping registers are (arrays of) registers of "your" type (both are read-write registers, one or the other is active depending on the operation mode of the peripheral).

With some 'try and error' I managed *) to extend the .svd2ada file to make it possible to process the complete SVD file:
```xml
<?xml version="1.0"?>
<svd2ada apply="LPC800_v0.3.svd">
  <peripheral name="CRC">
    <discriminent id="A" type="Access_Mode">
      <value>RO</value>
      <value>WO</value>
    </discriminent>
    <register name="SUM" when="RO"/>
    <register name="WR_DATA" when="WO"/>
  </peripheral>
  <peripheral name="SCT">
    <discriminent id="M" type="Mode">
      <value>Match</value>
      <value>Capture</value>
    </discriminent>
    <register name="MATCH" when="Match"/>
    <register name="CAP" when="Capture"/>
    <register name="MATCHREL" when="Match"/>
    <register name="CAPCTRL" when="Capture"/>
  </peripheral>
</svd2ada>
```
*) That shall not mean that a proper documentation for svd2ada is completely superfluous :wink:

lambourg, 2016-04-20:
The issue I see is the soundness of the generated binding. My goal here is to have svd2ada generate sound bindings, on which you can rely upon when they're generated. So clearly having fuzzy algorithm to try to determine whether two registers with very different naming scheme are the same (only one is write-only the other one being read-only) and automatically determine "where the difference is" is a clear issue. What if the algorithm leads to unsound binding? Where are the situations where the algorithm may fail? What about registers with both operation mode *and* RO/WO registers? Basically, I love explicit definition, and deal with implicit ones with great care.

Also I'm a reluctant to introduce MCU-specific code in svd2ada.

This being said, I can give it a try on determining if one register is fully read-only while the other is fully write-only (potentially without fields or with a single field?) and then automatically introduce a "Write_Mode". This is a bit complex (as I then need to consider the case where several modifiers can be applied to the peripheral).

Matthias-R, 2016-04-21:
Don't put too much effort in this! 

> Where are the situations where the algorithm may fail? What about registers with both operation mode and RO/WO registers?

That's an important point. Better have to do a little manual work than to have an automatism which tends to fail sometimes.
And - as I mentioned in my last post - this configuration occurs more seldom than I thought first. Maybe I'll revise my opinion if I stumble upon a controller which has really many of these RO/WO register pairs. But for now, I think, the possibilities to control the generated code with the .svd2ads file is sufficient.

lambourg, 2016-04-24:
Thanks for your feedback. I' closing this ticket then (still keeping in mind I'll have to write some documentation at some point of course ;) ).

closed 2016-04-24 by lambourg