AdaCore/svd2ada #14: Default value supplied for read-only array

simonjwright, 2016-04-20:
commit 215a0bb49f4c90aef2d96d634e0f10c896a28c06 prevented supplying a (useless) default value for read-only registers. A default value is still supplied for an _array_ component of a read-only register: from CMSIS-SVD/STM32F429x.svd,
```
   --  GPIO port input data register
   type IDR_Register is record
      --  Read-only. Port input data (y = 0..15)
      IDR            : IDR_Field := (As_Array => False, Val => 16#0#);
      --  unspecified
      Reserved_16_31 : Interfaces.Bit_Types.Short;
   end record
     with Volatile_Full_Access, Size => 32,
          Bit_Order => System.Low_Order_First;
```
(I spotted this because gnatprove objects to partial initialization).

lambourg, 2016-04-27:
Now that this has been partially reverted (rational being that most registers are only partly read-only and when this is the case then writing to a read-only field has no effect, so we still need the shortcut (others => <>) at the software level in such case.

So is it still the case that array components use improper default values? (to be honest, I haven't checked). If so, can you point me to a reproducer? (svd file and the register that defines such array).

simonjwright, 2016-04-27:
I think the inaccurate title I originally gave this issue was misleading. It’s not that the default value supplied is wrong, but that only half of this register has a default value, and SPARK doesn’t permit partial initialization.
I think that the chance of SPARK being asked to analyze the generated code is pretty remote - for a start, no subprograms! - and that being said, I’m closing the issue.

closed 2016-04-27 by simonjwright

lambourg, 2016-04-27:
Ok, thanks for the feedback!