AdaCore/svd2ada #16: Could warnings be added as comments in generated files?

simonjwright, 2016-04-25:
When `svd2ada` says
```
... Found overlapping registers: CCMR1_Output
... Found overriding registers: CCMR1_Input
... Found overlapping registers: CCMR2_Output
... Found overriding registers: CCMR2_Input
... you may check the automatic generation for those registers
```
could equivalent comments be added to the generated code?

lambourg, 2016-04-27:
Interesting idea. We could either generate a comment or event a pragma Compile_Time_Warning in such situation (although I hope that the default behavior of svd2ada is OK in most cases here).

Another option would be to propose means to create an *.svd2ada helper file that indicates to svd2ada how to handle this case.

I'll sleep a bit on the idea and will come back to you. Thanks for the feedback!

Matthias-R, 2016-05-18:
> Another option would be to propose means to create an *.svd2ada helper file that indicates to svd2ada how to handle this case.

Perhaps even a skeleton for the svd2ada helper file could be generated? Especially if svd2ada is not able to resolve the overlapping register definitions automatically.

lambourg, 2016-05-18:
Well, I have currently a design issue here: when automated resolution is OK, I create union types at the register level, while when the svd2ada file is in use, the union type is at the peripheral level.

I guess I'll have to fix that first before being able to merge anything and provide this kind of functionality.

Matthias-R, 2016-05-18:
> Well, I have currently a design issue here: when automated resolution is OK, I create union types at the register level, while when the svd2ada file is in use, the union type is at the peripheral level.

I would generally prefer the approach with the union at the peripheral level. The union at the register level adds an extra level in the type hierarchy (which has no correspondency in the datasheet for the device). The union at the peripheral level avoids this additional layer.

lambourg, 2016-05-18:
Agreed. That's why I need to find a way to summarize everything at the peripheral level, and not at the register level (where this is detected).

lambourg, 2016-05-22:
Some updates on this front: the proper support for registers being defined at the same address offset is now implemented in svd2ada, removing both the warnings and the need this kind of helper file.

I'm thus closing this issue and invite you to try out the new version :)

closed 2016-05-22 by lambourg

simonjwright, 2016-05-22:
Excellent! Thanks ...