AdaCore/svd2ada #17: Aliased registers in LPC43xx_43Sxx.svd

martinmarcos, 2016-04-29:
I am trying to port the Ada Runtime Library to NXP's LPC4337 MCU and I am following the instructions from the following blog post: [Porting the Ada Runtime to a new ARM board](http://blog.adacore.com/porting-the-ada-runtime-to-a-new-arm-board). The post suggests using **_svd2ada_** to get the list of interrupts. I download **_svd2ada_**, build it and ran it on the **_LPC43xx_43Sxx.svd_** file and it threw the following error:

```
*** WARNING: ignoring device element headerDefinitionsPrefix
*** WARNING ***: cannot find a proper Ada type name for the aliased registers MATCH and CAP

raised CONSTRAINT_ERROR : descriptors-register.adb:271 index check failed
```

_Matthias-R_ had a similar [issue](https://github.com/AdaCore/svd2ada/issues/12). So I am thinking I have to write a **_LPC43xx_43Sxx.svd2ada_** file to work around the aliased registers, is that correct? If so I'm not sure how to write it.

lambourg, 2016-04-30:
The svd2ada file expects the following elements:

```
<?xml version="1.0"?>
<svd2ada apply="[THE_SVD_FILE].svd">
  <peripheral name="[PERIPHERAL NAME AS NAMED IN SVD FILE]">
    <discriminent id="[THE ENUM IDENTIFIER THAT WILL DISCRIMINATE THE REGISTERS]" 
                           type="[THE ENUM TYPE]">
      <value>[VALUE1 OF THE ENUM]</value>
      <value>[VALUE2 OF THE ENUM</value>
      <!-- you can put any number of values as needed here -->
    </discriminent>
    <register name="[FIRST ALIASED REGISTER]" when="[VALUE OF THE ENUM THAT APPLIES"/>
    <register name="[SECOND ALIASED REGISTER]" when="[VALUE OF THE ENUM THAT APPLIES]"/>
  </peripheral>
</svd2ada>
```

so for example in your case it would be something like

```
<?xml version="1.0"?>
<svd2ada apply="LPC43xx_43Sxx.svd">
  <peripheral name="SGPIO0">
    <discriminent id="Mode" type="SGPIO_Mode">
      <value>Match_Mode</value>
      <value>Capture_Mode</value>
    </discriminent>
    <register name="MATCH" when="Match_Mode"/>
    <register name="CAP" when="Capture_Mode"/>
  </peripheral>
</svd2ada>
```

martinmarcos, 2016-04-30:
Ok. So I am kind of getting the idea of how to write the _.svd2ada_ file, but I still have doubts with what exactly I have to write for the _discriminent_ (by the way shouldn’t the word be _"discriminant"_?). It would certainly help if I was familiarized with the _svd-cmsis_ schema.

The peripheral is the **_SCT_** or _State Configurable Timer_, and I found the register that controls the MATCH/CAP mode of the peripheral.

```
<register>
     <name>REGMODE</name>
     <description>SCT match/capture registers mode register</description>
     <addressOffset>0x04C</addressOffset>
     <access>read-write</access>
     <resetValue>0x00000000</resetValue>
     <resetMask>0xFFFFFFFF</resetMask>
     <fields>
          <field>
               <name>REGMOD_L</name>
               <description>Each bit controls one pair of match/capture registers (register 0 = bit 0, register 1 = bit 1,..., register 15 = bit 15).  0 = registers operate as match registers. 1 = registers operate as capture registers.</description>
               <bitRange>[15:0]</bitRange>
          </field>
          <field>
               <name>REGMOD_H</name>
               <description>Each bit controls one pair of match/capture registers (register 0 = bit 16, register 1 = bit 17,..., register 15 = bit 31). 0 = registers operate as match registers. 1 = registers operate as capture registers.</description>
               <bitRange>[31:16]</bitRange>
          </field>
     </fields>
</register>
```

But I don't understand how to write the _discriment_ tag from that.

lambourg, 2016-05-12:
well, the discrimin*a*nt (sorry, will fix this) part is something that is not part of the SVD file, it's something you have to come up with by understanding the way the peripheral works and its modes.
That why svd2ada needs a user input for that: it's not in the SVD file explicitely.

Ada requires Union types to define a discriminant:
```
type Blah (Disc : Disc_Type) is record
   case Disc is
      when Foo =>
         Foo_Fields : Foo_Fields_Types;
      when Bar =>
         Bar_Fields : Bar_Fields_Types;
   end case;
end record with Unchecked_Union;

```
The discriminant will never be present into the assembler code, so has no real impact on the generated code. However it has great impact on how you understand the code.

lambourg, 2016-05-22:
Ok, some update: the helper file <name>.svd2ada is not needed anymore (I thus remove its handling by svd2ada). The peripheral should now be generated properly without the need to specify anything.

I'm thus closing this issue. Thanks for your report!

closed 2016-05-22 by lambourg