AdaCore/svd2ada #19: ADA.STRINGS.INDEX_ERROR while trying to generate from LPC43xx_43Sxx.svd

martinmarcos, 2016-06-21:
I am trying to generate Ada bindings from LPC43xx_43Sxx.svd and I am getting the following error:

```
martin@martin-notebook:~/software/lpcopen$ svd2ada -o /home/martin/dev/ada/lpc4337 LPC43xx_43Sxx.svd
Generate SCT
Generate GPDMA
Generate SPIFI
Generate SDMMC
Generate EMC
Generate USB0
Generate USB1
Generate LCD
Generate EEPROM
Generate ETHERNET
Generate ATIMER
Generate REGFILE
Generate PMC
Generate CREG
Generate EVENTROUTER
Generate RTC
Generate CGU
Generate CCU1
Generate CCU2
Generate RGU
Generate WWDT
Generate USART
Generate UART1
Generate SSP
Generate TIMER
Generate SCU
Generate GPIO_PIN_INT
Generate GPIO_GROUP_INT0
Generate MCPWM
Generate I2C
Generate I2S
Generate C_CAN1

raised ADA.STRINGS.INDEX_ERROR : a-strunb.adb:779
```

Regards
Martin

lambourg, 2016-06-21:
Hello Martin,

Is there a way I could download this SVD file? Alternatively a backtrace from a debug session would be helpful to determine where the issue is...

Regards,
- Jerome

martinmarcos, 2016-06-21:
Jerome,

You can download the svd file from [https://www.lpcware.com/content/nxpfile/lpc43sxx-svd-file](https://www.lpcware.com/content/nxpfile/lpc43sxx-svd-file). I never did a backtrace on linux before. Let me find out how to do it and I'll send it to you. If you could point out where I could learn how to do that I would greatly appreciate it.

Regards,
Martin

martinmarcos, 2016-06-21:
Did this:

```
(gdb) break exit
Breakpoint 6 at 0x7ffff7a511e0: file exit.c, line 104.
(gdb) run -o /home/martin/dev/ada/lpc4337 LPC43xx_43Sxx.svd
Starting program: /home/martin/bin/svd2ada/svd2ada -o /home/martin/dev/ada/lpc4337 LPC43xx_43Sxx.svd
Generate SCT
Generate GPDMA
Generate SPIFI
Generate SDMMC
Generate EMC
Generate USB0
Generate USB1
Generate LCD
Generate EEPROM
Generate ETHERNET
Generate ATIMER
Generate REGFILE
Generate PMC
Generate CREG
Generate EVENTROUTER
Generate RTC
Generate CGU
Generate CCU1
Generate CCU2
Generate RGU
Generate WWDT
Generate USART
Generate UART1
Generate SSP
Generate TIMER
Generate SCU
Generate GPIO_PIN_INT
Generate GPIO_GROUP_INT0
Generate MCPWM
Generate I2C
Generate I2S
Generate C_CAN1

raised ADA.STRINGS.INDEX_ERROR : a-strunb.adb:779

Breakpoint 6, __GI_exit (status=1) at exit.c:104
104	exit.c: No such file or directory.
(gdb) where
#0  __GI_exit (status=1) at exit.c:104
#1  0x00000000005c9e69 in __gnat_os_exit ()
#2  0x00000000005cba7d in __gnat_last_chance_handler ()
#3  0x000000000057fb2c in ada.exceptions.exception_traces.unhandled_exception_terminate (excep=0x1e4b0a0) at a-exextr.adb:163
#4  0x000000000057f98d in <__gnat_unhandled_except_handler> (
    gcc_exception=<optimized out>) at a-exexpr.adb:395
#5  0x00000000005cb9ac in __gnat_cleanupunwind_handler ()
#6  0x00000000005cf3a9 in _Unwind_ForcedUnwind_Phase2 (exc=0x1e4b060, 
    context=0x7fffffff8b10) at ../../../src/libgcc/unwind.inc:163
#7  0x00000000005cfac6 in _Unwind_Resume (exc=exc@entry=0x1e4b060)
    at ../../../src/libgcc/unwind.inc:232
#8  0x00000000004040f3 in svd2ada ()
    at /home/martin/software/gnat/svd2ada/src/svd2ada.adb:206
```
Tell me if this helps at all.

Regards.

lambourg, 2016-06-21:
basically, it's:
gdb svd2ada
(gdb) catch exception
(gdb) run -o /home/martin/dev/ada/lpc4337 LPC43xx_43Sxx.svd
<GDB breaks on the exception>
(gdb) bt
<backtrace is printed>

svd2ada needs to be built with debug information.

lambourg, 2016-06-21:
Breaking on exit is too late: it's after the exception is handled. Here I'd like to have the backtrace at the time the exception is raised.

martinmarcos, 2016-06-21:
I built svd2ada with default settings, so I think it was built with debug information. Its the "-g" option, right?

```
martin@martin-notebook:~/software/lpcopen$ gdb svd2ada 
GNU gdb (GDB) 7.8.90 for GNAT GPL 2015 [rev=gdb-head-ref-102395-gccbe2a7]
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
See your support agreement for details of warranty and support.
If you do not have a current support agreement, then there is absolutely
no warranty for this version of GDB.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from svd2ada...done.
(gdb) catch exception 
Catchpoint 1: all Ada exceptions
(gdb) run -o /home/martin/dev/ada/lpc4337 LPC43xx_43Sxx.svd
Starting program: /home/martin/bin/svd2ada/svd2ada -o /home/martin/dev/ada/lpc4337 LPC43xx_43Sxx.svd
Generate SCT
Generate GPDMA
Generate SPIFI
Generate SDMMC
Generate EMC
Generate USB0
Generate USB1
Generate LCD
Generate EEPROM
Generate ETHERNET
Generate ATIMER
Generate REGFILE
Generate PMC
Generate CREG
Generate EVENTROUTER
Generate RTC
Generate CGU
Generate CCU1
Generate CCU2
Generate RGU
Generate WWDT
Generate USART
Generate UART1
Generate SSP
Generate TIMER
Generate SCU
Generate GPIO_PIN_INT
Generate GPIO_GROUP_INT0
Generate MCPWM
Generate I2C
Generate I2S
Generate C_CAN1

Catchpoint 1, ADA.STRINGS.INDEX_ERROR at 0x00000000004d749a in base_types.common_prefix (name1=..., name2=...)
    at /home/martin/software/gnat/svd2ada/src/base_types.adb:618
618	        and then Element (Name2, Prefix) = '_'
(gdb) bt
#0  <__gnat_debug_raise_exception> (e=0x893680 <ada.strings.index_error>) at s-excdeb.adb:43
#1  0x00000000005803c7 in ada.exceptions.complete_occurrence (x=x@entry=0x1e4b0a0) at a-except.adb:924
#2  0x00000000005803d3 in ada.exceptions.complete_and_propagate_occurrence (x=x@entry=0x1e4b0a0) at a-except.adb:933
#3  0x0000000000580426 in <__gnat_raise_exception> (e=0x893680 <ada.strings.index_error>, message=...) at a-except.adb:975
#4  0x0000000000586747 in ada.strings.unbounded.element ()
#5  0x00000000004d749a in base_types.common_prefix (name1=..., name2=...) at /home/martin/software/gnat/svd2ada/src/base_types.adb:618
#6  0x00000000004a2d08 in descriptors.field.similar_field (first=2147483647, prefix_idx=10, f2=..., f1=...)
    at /home/martin/software/gnat/svd2ada/src/descriptors-field.adb:183
#7  descriptors.field.dump (spec=..., reg=reg@entry=0x1db3c80, rec=..., reg_fields=..., properties=...)
    at /home/martin/software/gnat/svd2ada/src/descriptors-field.adb:432
#8  0x0000000000491619 in descriptors.register.dump (spec=..., reg=0x1db3c80) at /home/martin/software/gnat/svd2ada/src/descriptors-register.adb:349
#9  0x00000000004837e7 in descriptors.peripheral.dump (group=..., dev_name=..., output_dir=...)
    at /home/martin/software/gnat/svd2ada/src/descriptors-peripheral.adb:535
#10 0x000000000046294e in descriptors.device.dump (device=..., output_dir=...) at /home/martin/software/gnat/svd2ada/src/descriptors-device.adb:475
#11 0x0000000000403fc7 in svd2ada () at /home/martin/software/gnat/svd2ada/src/svd2ada.adb:187
```

lambourg, 2016-06-21:
Thanks! This shows us that the issue is in base_types.adb:618. I'll investigate and will come back to you when the bug is fixed (or when I have more explanations).

lambourg, 2016-06-21:
And indeed, -g is responsible for adding the debug informations.

nocko, 2016-06-30:
Another SVD that fails with the same exception:
https://gist.github.com/nocko/aacdb51b31cc59cafe57b9c653255895

nocko, 2016-06-30:
Catchpoint 1, ADA.STRINGS.INDEX_ERROR at 0x0000000000467bd6 in descriptors.peripheral.find_overlapping_registers (reg_set=...)
    at /home/nock/devel/svd2ada/src/descriptors-peripheral.adb:271
271                          if Prefix (J) /= Element (Reg_Set (K).Name, J) then
(gdb) bt
#0  <__gnat_debug_raise_exception> (e=0x8983c0 <ada.strings.index_error>, message=...) at s-excdeb.adb:40
#1  0x000000000058de37 in ada.exceptions.complete_occurrence (x=x@entry=0x1183270) at a-except.adb:925
#2  0x000000000058de46 in ada.exceptions.complete_and_propagate_occurrence (x=x@entry=0x1183270) at a-except.adb:936
#3  0x000000000058de97 in <__gnat_raise_exception> (e=0x8983c0 <ada.strings.index_error>, message=...) at a-except.adb:978
#4  0x000000000059408a in ada.strings.unbounded.element (source=..., index=<optimized out>) at a-strunb.adb:779
#5  0x0000000000467bd6 in descriptors.peripheral.find_overlapping_registers (reg_set=...)
    at /home/nock/devel/svd2ada/src/descriptors-peripheral.adb:271
#6  0x0000000000473077 in descriptors.peripheral.dump_periph_type.create_record ()
    at /home/nock/devel/svd2ada/src/descriptors-peripheral.adb:387
#7  descriptors.peripheral.dump_periph_type (type_name=...) at /home/nock/devel/svd2ada/src/descriptors-peripheral.adb:407
#8  0x000000000047a557 in descriptors.peripheral.dump (group=..., dev_name=..., output_dir=...)
    at /home/nock/devel/svd2ada/src/descriptors-peripheral.adb:563
#9  0x000000000045c1c7 in descriptors.device.dump (device=..., output_dir=...)
    at /home/nock/devel/svd2ada/src/descriptors-device.adb:475
#10 0x000000000040807c in svd2ada () at /home/nock/devel/svd2ada/src/svd2ada.adb:187

nocko, 2016-06-30:
Removing the following section of the SVD causes the program to complete successfully.

```
 <register>
          <dim>4</dim>
          <dimIncrement>0x04</dimIncrement>
          <dimIndex>0-3</dimIndex>
          <name>SIZERAMBLOCK[%s]</name>
          <description>Deprecated array of size of RAM block in bytes. This name is kept for backward compatinility purposes. Use SIZERAMBLOCKS instead.</description>
          <addressOffset>0x038</addressOffset>
          <alternateRegister>SIZERAMBLOCKS</alternateRegister>
        </register>
```

lambourg, 2016-06-30:
Thank you very much for digging into this... Unfortunately I won't have much time to work on this in the coming days, but I'll keep you in touch as soon as this is fixed. Of course, if you have a patch, I'll be more than happy to apply it.

However, from the look of it, it seems safe to just remove the definition from the SVD file as it's deemed "deprecated".

nocko, 2016-06-30:
I am quite new to Ada, but will look into submitting a patch.

martinmarcos, 2016-06-30:
@nocko, I think the exception is the same, but the issue is different. The backtraces show the exception is raised in different files.

@lambourg, I am new to Ada as well, but I'll try to work a patch for the issue I've encountered. I've looked at the function that causes the problem, and I kind got an idea of why it is failing. If I get to work I'll send you a Pull Request.

Regards.

simonjwright, 2016-07-02:
In both cases the problem is comparing two strings to find a common prefix where the second string is shorter.

The attached patch gets rid of both exceptions.

There are remaining problems in that both problematic SVDs generate code that doesn’t compile (the LPC one more so :-)) - it may be that the patch doesn’t go far enough, or something else.

[issue19.zip](https://github.com/AdaCore/svd2ada/files/344426/issue19.zip)

lambourg, 2016-07-02:
SImon, I've merged your change, thank you!

What's the compilation error on the LPC svd file?

Matthias-R, 2016-07-02:
> What's the compilation error on the LPC svd file?

LPC800.svd contains enumerations with duplicate elements. Apparently, they simply have reused the first 20 characters of the description string as identifier for the enumeration elements. Sometimes, these first 20 characters are identical in multiple descriptions...

Additional, some enumeration identifiers are not legal Ada identifiers. Some of them have consecutive underscores or a underscore at the end. Others start with a number - as far as I have seen, you already have a workaround for this oddity.

Has LPC43xx_43Sxx.svd the same issues?

Regarding the problem with the duplicates, I think that the converter cannot do anything useful here. You may make the identifiers unique by adding a prefix or postfix - but the resulting names are not really meaningful. The only way to get useful names is to fix the svd file.

simonjwright, 2016-07-02:
I think the LPC problems are really to do with the SVD file.

- There are many names with trailing underscores
- A lot (all?) of those names are RESERVED_, which are (a) illegal, (b) often duplicated, (c) unnecessary since SVD2Ada sorts that out automatically
- I think a tool was used to generate the SVD, because the names are truncated to a max length e.g. “Rising_Edges_On_Inpu” even where that results in duplicate names

After that, there are what may be real problems in SVD2Ada:

```
lpc43xx-rgu.ads:1505:28: component clause not allowed for variable length component
```

simonjwright, 2016-07-02:
The nrf51 file that @nocko uploaded has a problem in PPI/CHEN, which has bits CH0, CH1 etc to CH31, and for some reason SVD2Ada fails to identify the complete range of bits so it ends up with several arrays all called CH.
```
   --  Channel enable.
   type CHEN_Register is record
      --  Enable PPI channel 0.
      CH             : CHEN_CH_Field := (As_Array => False, Val => 16#0#);
      --  Enable PPI channel 3.
      CH3            : CH3_Field := Disabled;
      --  Enable PPI channel 4.
      CH             : CHEN_CH_Field_1 := (As_Array => False, Val => 16#0#);
      --  unspecified
      Reserved_16_19 : nrf51.UInt4 := 16#0#;
      --  Enable PPI channel 20.
      CH             : CHEN_CH_Field_2 := (As_Array => False, Val => 16#0#);
   end record
     with Volatile_Full_Access, Size => 32,
          Bit_Order => System.Low_Order_First;
```

lambourg, 2016-07-11:
The LPC bindings should now be properly generated: both the '1x' enum value is now handled and the issue with duplicated enum ids are now fixed in svd2ada. So I'm going to close this issue, feel free to open a new one if other issues arise.

closed 2016-07-11 by lambourg