AdaCore/svd2ada #25: Error in generated code: component overlaps in record

euripedesrocha, 2016-07-21:
I' trying to generate code for [atsamd4ls](ds.arm.com/media/resources/db/chip/atmel/atsam4ls2a/ATSAM4LS2A.svd). The code was generated with the svd file in the link using the following command without any issues:

svd2ada -p Interfaces.ATSAM4LS -o bsp

The build gives me the follwing problem:

i-atsam4ls-gpio.ads:2911:07: component "GPERS" overlaps "GPER" at line 2910

All fields in the record have the same problem. 

```ada
   for GPIO_Peripheral use record
      GPER      at 0 range 0 .. 12287;
      GPERS     at 4 range 0 .. 12287;
      GPERC     at 8 range 0 .. 12287;
      GPERT     at 12 range 0 .. 12287;
...
  end record
```
I'm quite new to Ada, but it semms that the register is longer than it should be. I tried to change the sizes to 32 and it fixes the problem for a few lines:

```ada 
     GPER  at 0 range 0 .. 31
```
The error changes to a minimum size issue. I guess some registers are missing.

euripedesrocha, 2016-07-22:
Working more on the issue, It seems to be the same case of Issue #5 .

lambourg, 2016-07-24:
This indeed looks the same, and using the latest version of svd2ada, the generation is correct:

`   for GPIO_Peripheral use record
      GPER0      at 0 range 0 .. 31;
      GPERS0     at 4 range 0 .. 31;
      GPERC0     at 8 range 0 .. 31;
      GPERT0     at 12 range 0 .. 31;
      PMR00      at 16 range 0 .. 31;
      PMR0S0     at 20 range 0 .. 31;
      PMR0C0     at 24 range 0 .. 31;
      PMR0T0     at 28 range 0 .. 31;
      PMR10      at 32 range 0 .. 31;
      PMR1S0     at 36 range 0 .. 31;
      PMR1C0     at 40 range 0 .. 31;
      PMR1T0     at 44 range 0 .. 31;
      PMR20      at 48 range 0 .. 31;
      PMR2S0     at 52 range 0 .. 31;
      PMR2C0     at 56 range 0 .. 31;
      PMR2T0     at 60 range 0 .. 31;
      ODER0      at 64 range 0 .. 31;
      ODERS0     at 68 range 0 .. 31;
      ODERC0     at 72 range 0 .. 31;
      ODERT0     at 76 range 0 .. 31;
      OVR0       at 80 range 0 .. 31;
      OVRS0      at 84 range 0 .. 31;
      OVRC0      at 88 range 0 .. 31;
      OVRT0      at 92 range 0 .. 31;
      PVR0       at 96 range 0 .. 31;
      PUER0      at 112 range 0 .. 31;
      PUERS0     at 116 range 0 .. 31;
      PUERC0     at 120 range 0 .. 31;
      PUERT0     at 124 range 0 .. 31;
      PDER0      at 128 range 0 .. 31;
      PDERS0     at 132 range 0 .. 31;
      PDERC0     at 136 range 0 .. 31;
      PDERT0     at 140 range 0 .. 31;
      IER0       at 144 range 0 .. 31;
      IERS0      at 148 range 0 .. 31;
      IERC0      at 152 range 0 .. 31;
      IERT0      at 156 range 0 .. 31;
      IMR00      at 160 range 0 .. 31;
      IMR0S0     at 164 range 0 .. 31;
      IMR0C0     at 168 range 0 .. 31;
      IMR0T0     at 172 range 0 .. 31;
      IMR10      at 176 range 0 .. 31;
      IMR1S0     at 180 range 0 .. 31;
      IMR1C0     at 184 range 0 .. 31;
      IMR1T0     at 188 range 0 .. 31;
      GFER0      at 192 range 0 .. 31;
      GFERS0     at 196 range 0 .. 31;
      GFERC0     at 200 range 0 .. 31;
      GFERT0     at 204 range 0 .. 31;
      IFR0       at 208 range 0 .. 31;
      IFRC0      at 216 range 0 .. 31;
      ODMER0     at 224 range 0 .. 31;
      ODMERS0    at 228 range 0 .. 31;
      ODMERC0    at 232 range 0 .. 31;
      ODMERT0    at 236 range 0 .. 31;
      ODCR00     at 256 range 0 .. 31;
      ODCR0S0    at 260 range 0 .. 31;
      ODCR0C0    at 264 range 0 .. 31;
      ODCR0T0    at 268 range 0 .. 31;
      ODCR10     at 272 range 0 .. 31;
      ODCR1S0    at 276 range 0 .. 31;
      ODCR1C0    at 280 range 0 .. 31;
      ODCR1T0    at 284 range 0 .. 31;
      OSRR00     at 304 range 0 .. 31;
      OSRR0S0    at 308 range 0 .. 31;
      OSRR0C0    at 312 range 0 .. 31;
      OSRR0T0    at 316 range 0 .. 31;
      STER0      at 352 range 0 .. 31;
      STERS0     at 356 range 0 .. 31;
      STERC0     at 360 range 0 .. 31;
      STERT0     at 364 range 0 .. 31;
      EVER0      at 384 range 0 .. 31;
      EVERS0     at 388 range 0 .. 31;
      EVERC0     at 392 range 0 .. 31;
      EVERT0     at 396 range 0 .. 31;
      LOCK0      at 416 range 0 .. 31;
      LOCKS0     at 420 range 0 .. 31;
      LOCKC0     at 424 range 0 .. 31;
      LOCKT0     at 428 range 0 .. 31;
      UNLOCK0    at 480 range 0 .. 31;
      ASR0       at 484 range 0 .. 31;
      PARAMETER0 at 504 range 0 .. 31;
      VERSION0   at 508 range 0 .. 31;
      GPER1      at 512 range 0 .. 31;
`

Can you check that you're up-to-date?

euripedesrocha, 2016-07-24:
You're correct. The most recent version doesn't have the issue. It was my fault. I forgot that I forked the repo, and this lead me to think that the code was up to date.

closed 2016-07-24 by euripedesrocha