AdaCore/svd2ada #27: Code generated for overlapping "dim" groupings of registers is wrong

jgrivera67, 2016-07-31:
The code generated by svd2ada for overlapping "dim" groupings of registers like
the following example  is wrong:
```
        <register>
          <dim>6</dim>
          <dimIncrement>0x8</dimIncrement>
          <dimIndex>0,1,2,3,4,5</dimIndex>
          <name>C%sSC</name>
          <description>Channel (n) Status and Control</description>
          **<addressOffset>0xC</addressOffset>**
          <size>32</size>
          <access>read-write</access>
          <resetValue>0</resetValue>
          <resetMask>0xFFFFFFFF</resetMask>
          <fields>
            ...
          </fields>
        </register>
        <register>
          <dim>6</dim>
          <dimIncrement>0x8</dimIncrement>
          <dimIndex>0,1,2,3,4,5</dimIndex>
          <name>C%sV</name>
          <description>Channel (n) Value</description>
          **<addressOffset>0x10</addressOffset>**
          <size>32</size>
          <access>read-write</access>
          <resetValue>0</resetValue>
          <resetMask>0xFFFFFFFF</resetMask>
          <fields>
          ...
          </fields>
        </register>
```
The code generated by svd2ada looks like this:
```
   --  Channel (n) Status and Control
   type TPM0_CSC_Registers is array (0 .. 5) of TPM0_CSC_Register;
   ...
   --  Channel (n) Value
   type TPM0_CV_Registers is array (0 .. 5) of TPM0_CV_Register;
   ...
   --  Timer/PWM Module
   type TPM0_Peripheral is record
      --  Status and Control
      SC     : TPM0_SC_Register;
      --  Counter
      CNT    : TPM0_CNT_Register;
      --  Modulo
      MOD_k  : TPM0_MOD_Register;
      --  Channel (n) Status and Control
      **CSC    : TPM0_CSC_Registers;**
      --  Channel (n) Value
      **CV     : TPM0_CV_Registers;**
      --  Capture and Compare Status
      STATUS : TPM0_STATUS_Register;
      --  Configuration
      CONF   : TPM0_CONF_Register;
   end record
     with Volatile;

   for TPM0_Peripheral use record
      SC     at 0 range 0 .. 31;
      CNT    at 4 range 0 .. 31;
      MOD_k  at 8 range 0 .. 31;
      **CSC    at 12 range 0 .. 383;**
      **CV     at 16 range 0 .. 383;**
      STATUS at 80 range 0 .. 31;
      CONF   at 132 range 0 .. 31;
   end record;
```
The **CSC** and **CV** arrays will overlap in the physical memory space, which is wrong. Indeed,
this code does not compile.
svd2ada should not be generating Ada arrays for them.  In contrast, the ARM SVDConv tool generates
the following C code for the same SVD fragment:
```
`typedef struct {                                    /*!< TPM0 Structure                                                        */
     __IO uint32_t  SC;                                /*!< Status and Control                                                    */
     __IO uint32_t  CNT;                               /*!< Counter                                                               */
     __IO uint32_t  MOD;                               /*!< Modulo                                                                */
     __IO uint32_t  C0SC;                              /*!< Channel (n) Status and Control                                        */
     __IO uint32_t  C0V;                               /*!< Channel (n) Value                                                     */
     __IO uint32_t  C1SC;                              /*!< Channel (n) Status and Control                                        */
     __IO uint32_t  C1V;                               /*!< Channel (n) Value                                                     */
     __IO uint32_t  C2SC;                              /*!< Channel (n) Status and Control                                        */
     __IO uint32_t  C2V;                               /*!< Channel (n) Value                                                     */
     __IO uint32_t  C3SC;                              /*!< Channel (n) Status and Control                                        */
     __IO uint32_t  C3V;                               /*!< Channel (n) Value                                                     */
     __IO uint32_t  C4SC;                              /*!< Channel (n) Status and Control                                        */
     __IO uint32_t  C4V;                               /*!< Channel (n) Value                                                     */
     __IO uint32_t  C5SC;                              /*!< Channel (n) Status and Control                                        */
     __IO uint32_t  C5V;                               /*!< Channel (n) Value                                                     */
     __I  uint32_t  RESERVED[5];
     __IO uint32_t  STATUS;                            /*!< Capture and Compare Status                                            */
     __I  uint32_t  RESERVED1[12];
     __IO uint32_t  CONF;                              /*!< Configuration                                                         */
} TPM0_Type;`
```

As you can see the **CxSC** and **CxV** registers are interleaved in the C struct. svd2ada should generate similar code in Ada, as there is no construct in Ada to interleave elements of two arrays, as far as I know.

simonjwright, 2016-07-31:
What the SVD is trying to describe is an array of two-element records. What MCU is this for?

Is it the case that the aaa%sbbb format is only ever used in cases like this? if so, we could give up on trying to make an array and fall back on the C style.

jgrivera67, 2016-07-31:
This is for Freescale Kinetis micrcontrollers. For example,  the MKL25Z4 (see https://github.com/posborne/cmsis-svd/blob/master/data/Freescale/MKL25Z4.svd).

A better way of expressing an array of a grouping of registers in SVD syntax is to use clusters. Indeed, I reported the issue https://github.com/posborne/cmsis-svd/issues/33 for the cmsis-svd Github repo, in which I describe how to do it.

lambourg, 2016-08-09:
@jgrivera67 what version of svd2ada are you using? This is supposed to be already fixed by the commit 05d10c25266dd9f99b33230f856bc84e7019c98c. Also the support for clusters has been finally implemented today (see ea87ef70b2e1ef9a2fabdf06b6fab9510a3b6610).

closed 2016-08-09 by lambourg

jgrivera67, 2016-08-09:
I was using version 143a2f55b553f394561fa6ef6e999aeeff16fbd3. I'll upate to
the latest version.

Thanks,

German