AdaCore/svd2ada #28: svd2ada does not handle the cluster SVD construct correctly

jgrivera67, 2016-07-31:
The**cluster** SVD construct does not seem to be supported by the svd2ada tool.
a record type declaration for the cluster is missing in the generated code.
For example, for the following SVD fragment:
```
          ...
         <cluster>
            <dim>6</dim>
            <dimIncrement>8</dimIncrement>
            <dimIndex>0-5</dimIndex>
            <name>CHANNELS[%s]</name>
            <description>Grouping of Channels</description>
            <addressOffset>0xC</addressOffset>
            <register>
              <name>CnSC</name>
              <description>Channel (n) Status and Control</description>
              <addressOffset>0x0</addressOffset>
              <size>32</size>
              <access>read-write</access>
              <resetValue>0</resetValue>
              <resetMask>0xFFFFFFFF</resetMask>
              <fields>
               ...
              </fields>
            </register>
            <register>
              <name>CnV</name>
              <description>Channel (n) Value</description>
              <addressOffset>0x4</addressOffset>
              <size>32</size>
              <access>read-write</access>
              <resetValue>0</resetValue>
              <resetMask>0xFFFFFFFF</resetMask>
              <fields>
              ...
              </fields>
            </register>
        </cluster>
        ...
```
The code generated by the svd2ada tool looks like this:
```
   --  Grouping of Channels
   type TPM0_CHANNELS_Registers is array (0 .. 5) of MKL25Z4.UInt110926248;
   ...
   type TPM0_Peripheral is record
      ...
      --  Grouping of Channels
      CHANNELS : TPM0_CHANNELS_Registers;
     ...
   end record
     with Volatile;

   for TPM0_Peripheral use record
      ...
      CHANNELS at 12 range 0 .. 383;
      ...
   end record;
```

The reference to type `MKL25Z4.UInt110926248` in the declaration of `TPM0_CHANNELS_Registers`  seems to be a bug, since it is not declared anywhere. Besides, instead there should be a record type
declaration like the following:
```
type TPM0_CHANNEL_Type is record
      --  Channel (n) Status and Control
      CSC    :  TPM0_CSC_Register;
      --  Channel (n) Value
      CV     :  TPM0_CV_Register;
end record;
```
to be used in the declaration of TPM0_CHANNELS_Registers, like this:
```
type TPM0_CHANNELS_Registers is array (0 .. 5) of TPM0_CHANNEL_Type;
```

lambourg, 2016-08-09:
This is now implemented in ea87ef70b2e1ef9a2fabdf06b6fab9510a3b6610.

Best regards,
Jerome

jgrivera67, 2016-08-09:
> This is now implemented in ea87ef7
>
> Best regards,
> Jerome

This does not work as expected. I just tried the test case described in #28
<https://github.com/AdaCore/svd2ada/issues/28>,
and the code generated by the latest version of sv2ada looks like this:

   --  Grouping of Channels
   type TPM0_CHANNELS_Cluster is record
      --  Channel (n) Status and Control
      CnSC : TPM0_CnSCCHANNELS_Register;
      --  Channel (n) Value
      CnV  : TPM0_CnVCHANNELS_Register;
   end record
     with Volatile, Size => 64;

   for TPM0_CHANNELS_Cluster use record
      CnSC at 0 range 0 .. 31;
      CnV  at 4 range 0 .. 31;
   end record;
   ...

   --  Timer/PWM Module
   type TPM0_Peripheral is record
      --  Status and Control
      SC       : TPM0_SC_Register;
      --  Counter
      CNT      : TPM0_CNT_Register;
      --  Modulo
      MOD_k    : TPM0_MOD_Register;
      --  Grouping of Channels
      CHANNELS : TPM0_CHANNELS_Cluster;
      --  Capture and Compare Status
      STATUS   : TPM0_STATUS_Register;
      --  Configuration
      CONF     : TPM0_CONF_Register;
   end record
     with Volatile;

This is wrong, because CHANNELS should be an array
of clusters not just one cluster. The correct code
should look like this:

 --  One Channel
   type TPM0_CHANNELS_Cluster is record
      --  Channel (n) Status and Control
      CnSC : TPM0_CnSCCHANNELS_Register;
      --  Channel (n) Value
      CnV  : TPM0_CnVCHANNELS_Register;
   end record
     with Volatile, Size => 64;
 ...

 --  Grouping of Channels
   type TPM0_CHANNELS_Registers is array (0 .. 5) of TPM0_CHANNELS_Cluster;
   ...
   type TPM0_Peripheral is record
      ...
      --  Grouping of Channels
      CHANNELS : TPM0_CHANNELS_Registers;
     ...
   end record
     with Volatile;
   ...
   for TPM0_Peripheral use record
      ...
      CHANNELS at 12 range 0 .. 383;
      ...
   end record;

Thanks,

German

lambourg, 2016-08-09:
Interesting, indeed I missed the array of cluster case.

Ok, back to work then.

lambourg, 2016-08-09:
Can you send a complete reproducer svd file? Just so that I can test the generation of such construction.

lambourg, 2016-08-09:
This should now be fine with commit 6216627. Can you re-try?

closed 2016-08-09 by lambourg

jgrivera67, 2016-08-09:
I just tried and it works now.

Thanks,

German