AdaCore/svd2ada #30: how to generate handler.S and stm32_svd-interrupts.ads

derirremitderwaffe, 2017-02-05:
hi, I'd like to add a runtime for the stmf030 mcu. On the Adacore blog http://blog.adacore.com/porting-the-ada-runtime-to-a-new-arm-board it is said to use svd2ada to generate Handler.S and stm32_svd-interrupts.ads,also the files of the stm32f4/7 runtimes seem to be from svd2ada. But I don't get any. If I use svd2ada with STM32F030.svd, I just get the root and devices .ads files.

Fabien-Chouteau, 2017-02-05:
Hi,

You have to specify a root package name that starts with "Interfaces.".

For example:
`./svd2ada -p Interfaces.STM32 -o out CMSIS-SVD/ST/STM32F40x.svd`

I will keep this issue open as an enhancement request to start a documentation or add this info in the help message.

derirremitderwaffe, 2017-02-08:
Thank you, it worked

closed 2018-01-13 by lambourg