AdaCore/svd2ada #39: svd2ada build problems

kenitraskr, 2017-03-25:
Hello,
I have a lot problem for build svd2ada, i use ubuntu 32bits.
The problem is :
GNAT.Command_Line.Define_Switch error "svd2ada.adb:93:21: no candidate interpretations match the actuals"
And
svd2ada.adb:141:27: "Try_Help" not declared in "Command_Line"
Thank you for your help

lambourg, 2017-03-28:
Hello,

I suspect that the version of GNAT you're using does not contain the proper GNAT.Command_Line package. Have you tried with GNAT GPL from libre.adacore.com?

Best regards,
- Jerome

kenitraskr, 2017-04-01:
Hello Jerome,

Thanks for your reply, I have tried but I have a problem with the xml lib and
GNAT.Command_Line package.
Do you know how to install the xml library and GNAT.Command_Line package
with GNAT GPL from libre.adacore.com?

Here are the messages I have with :

svd2ada.adb:21:17: warning: unit "Ada.Strings.Unbounded" is not referenced
svd2ada.adb:21:39: missing "with Ada.Strings;"
svd2ada.adb:24:06: file "g-comlin.ads" not found
svd2ada.adb:24:10: warning: unit "GNAT.Command_Line" is not referenced
svd2ada.adb:25:06: file "g-dirope.ads" not found
svd2ada.adb:25:10: warning: unit "GNAT.Directory_Operations" is not referenced
svd2ada.adb:26:06: file "g-os_lib.ads" not found
svd2ada.adb:26:10: warning: unit "GNAT.OS_Lib" is not referenced
svd2ada.adb:27:06: file "g-string.ads" not found
svd2ada.adb:27:10: warning: unit "GNAT.Strings" is not referenced
svd2ada.adb:31:06: "Ada.Finalization" is not a predefined library unit
svd2ada.adb:31:06: "Svd2ada (body)" depends on "Sax.Readers (spec)"
svd2ada.adb:31:06: "Sax.Readers (spec)" depends on "Sax.Locators (spec)"
svd2ada.adb:31:06: "Sax.Locators (spec)" depends on "Sax.Symbols (spec)"
svd2ada.adb:31:06: "Sax.Symbols (spec)" depends on "Sax.Pointers (spec)"
svd2ada.adb:31:06: "Sax.Pointers (spec)" depends on "Ada.Finalization (spec)"
svd2ada.adb:32:06: file "g-dyntab.ads" not found
svd2ada.adb:32:06: "Schema.Validators (spec)" depends on
"Gnat.Dynamic_Tables (spec)"
svd2ada.adb:32:06: "Svd2ada (body)" depends on "Schema.Dom_Readers (spec)"
svd2ada.adb:32:06: "Schema.Dom_Readers (spec)" depends on
"Schema.Readers (spec)"
svd2ada.adb:32:06: "Schema.Readers (spec)" depends on "Schema.Validators (spec)"
svd2ada.adb:32:06: "Schema.Validators (spec)" depends on
"Schema.Simple_Types (spec)"
svd2ada.adb:32:06: "Schema.Simple_Types (spec)" depends on "Gnat.Regpat (spec)"
svd2ada.adb:33:06: file "g-dynhta.ads" not found
svd2ada.adb:33:06: "Svd2ada (body)" depends on "Schema.Schema_Readers (spec)"
svd2ada.adb:33:06: "Schema.Schema_Readers (spec)" depends on
"Gnat.Dynamic_Htables (spec)"
svd2ada.adb:38:06: "Base_Types (spec)" depends on "Ada.Containers (spec)"
svd2ada.adb:38:06: "Ada_Gen (spec)" depends on "Base_Types (spec)"
svd2ada.adb:38:06: "Base_Types (spec)" depends on "Ada.Strings (spec)"
svd2ada.adb:38:06: "Ada.Containers" is not a predefined library unit
svd2ada.adb:38:06: "Ada_Gen (spec)" depends on "Ada.Containers (spec)"
svd2ada.adb:38:06: "Ada.Strings" is not a predefined library unit
svd2ada.adb:38:06: "Svd2ada (body)" depends on "Ada_Gen (spec)"
svd2ada.adb:38:06: "Ada_Gen (spec)" depends on "Ada.Strings (spec)"
svd2ada.adb:39:06: "Ada.Strings" is not a predefined library unit
svd2ada.adb:39:06: "Descriptors.Device (spec)" depends on "Ada.Strings (spec)"
svd2ada.adb:39:06: "Descriptors.Peripheral (spec)" depends on
"Ada.Containers (spec)"
svd2ada.adb:39:06: "Descriptors.Peripheral (spec)" depends on
"Ada.Strings (spec)"
svd2ada.adb:39:06: "Descriptors.Cluster (spec)" depends on
"Ada.Containers (spec)"
svd2ada.adb:39:06: "Descriptors.Cluster (spec)" depends on "Ada.Strings (spec)"
svd2ada.adb:39:06: "Descriptors.Register (spec)" depends on
"Ada.Containers (spec)"
svd2ada.adb:39:06: "Descriptors.Register (spec)" depends on "Ada.Strings (spec)"
svd2ada.adb:39:06: "Descriptors.Field (spec)" depends on "Ada.Containers (spec)"
svd2ada.adb:39:06: "Descriptors.Field (spec)" depends on "Ada.Strings (spec)"
svd2ada.adb:39:06: "Descriptors.Enumerate (spec)" depends on
"Ada.Containers (spec)"
svd2ada.adb:39:06: "Svd2ada (body)" depends on "Descriptors.Device (spec)"
svd2ada.adb:39:06: "Descriptors.Device (spec)" depends on
"Descriptors.Peripheral (spec)"
svd2ada.adb:39:06: "Descriptors.Peripheral (spec)" depends on
"Descriptors.Cluster (spec)"
svd2ada.adb:39:06: "Descriptors.Cluster (spec)" depends on
"Descriptors.Register (spec)"
[... 64 lines cut ...]
ada_gen.ads:512:37: missing "with Ada.Containers;"
ada_gen.ads:515:07: missing "with Ada.Strings;"
ada_gen.ads:515:18: instantiation abandoned
ada_gen.ads:522:22: "Record_Field_Maps" is undefined
ada_gen.ads:527:28: "File_Type" not declared in "Text_IO"
ada_gen.ads:539:28: "File_Type" not declared in "Text_IO"
ada_gen.ads:551:28: "File_Type" not declared in "Text_IO"
ada_gen.ads:558:23: "With_Maps" is undefined
ada_gen.ads:561:23: "Element_Vectors" is undefined
base_types.ads:142:21: "Unbounded" is undefined (more references follow)
descriptors-peripheral.ads:53:25: "Address_Block_Vectors" is undefined
descriptors-peripheral.ads:54:25: "Interrupt_Vectors" is undefined
descriptors-peripheral.ads:80:20: "Peripheral_Vectors" is undefined
(more references follow)
descriptors-cluster.ads:66:17: "Peripheral_Element_Vectors" is
undefined (more references follow)
descriptors-register.ads:41:27: "Type_Holders" is undefined (more
references follow)
descriptors-field.ads:44:47: "Enumerate_Vectors" not declared in "Enumerate"
descriptors-field.ads:62:24: "Field_Vectors" is undefined (more
references follow)
descriptors-enumerate.ads:41:16: "Enumerate_Values_Vectors" is undefined
descriptors-enumerate.ads:50:20: "Enumerate_Vectors" is undefined
gprbuild: *** compilation phase failed
[2017-04-01 11:58:29] process exited with status 4, elapsed time: 00.67s

lambourg, 2017-04-01:
It looks as if you have no runtime at all with your compiler.

There must be an issue with your installation. At this point I would suggest to restart the GNAT GPL installation from the beginning (so delete the old compiler, download from the libre site, run doinstall and re-install in the installation directory of your choice.

Just in case there is a doubt: svd2ada is a native application, so in your case the native linux version of the compiler has to be installed.

Finally, svd2ada has to be compiled from its project file:

gprbuild -P svd2ada.

I hope that with the above instructions, you'll manage to achieve a successful build!

Best regards,
- Jerome

> Le 1 avr. 2017 à 12:33, kenitraskr <notifications@github.com> a écrit :
> 
> Hello Jerome,
> 
> Thanks for your reply, I have tried but I have a problem with the xml lib and
> GNAT.Command_Line package.

kenitraskr, 2017-04-02:
Hello Jerome,
I deleted the old compiler and i run doinstall.
Now i have this message:
Error: no native compiler found for language 'ada', default runtime
Thank you for your reply.

SaId

lambourg, 2018-02-11:
Have you solved your issue with installing the native GNAT compiler?

closed 2018-09-17 by lambourg

MilliUser, 2022-09-16:
Hello. I'm so new to ADA I'm a novice at some places. I found answers to some of my problems on the internet, but I couldn't get past the installation problem in svd2ada. I don't know if it is due to the file directory or some other problem.

The errors I get are after " gprbuild -P svd2ada.gpr ".
PS C:\GNAT\svd2ada-master> gprbuild -P svd2ada.gpr
Compile
   [Ada] svd2ada.adb
   [Ada] ada_gen.adb
   [Island] descriptors.ads
   [Ada] descriptors-device.adb
   [Ada] svd2ada_utils.adb
   [Ada] base_types.adb
   [Ada] base_types-register_properties.adb
   [Island] descriptors-peripheral.adb
   [Ada] descriptors-cluster.adb
   [Ada] descriptors-register.adb
   [Island] descriptors-field.adb
   [Ada] descriptors-enumerate.adb
bind
   [gprbind] svd2ada.bexch
   [Ada] svd2ada.ali
error: "s-imglld.ali" not found, "s-imglld.adb" must be compiled
error: "s-valenu.ali" not found, "s-valenu.adb" must be compiled
error: "dom.ads" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "dom-core.adb" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "sax.ads" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "sax-encodings.ads" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "unicode.adb" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "unicode-names.ads" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "unicode-names-basic_latin.ads" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "unicode-ces.adb" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "unicode-ces-utf32.adb" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12.....
it goes on and on...