AdaCore/svd2ada #45: Suboptimal code generation

LordPython, 2018-02-11:
While experimenting getting ada up and running on my STM32F2 nucleo board, I noticed that the machine code generated by using the svd2ada generated interfaces is sometimes rather suboptimal. For instance, as part of setting up one of the timers, I was doing the following:

    TIM4_Periph.CR1 := (
        CMS => Count_Mode_Up,
        DIR => Edge_Aligned,
        others => <>
    );

I thought this would be a good way to set the entire register to a known state during my startup code, while being able to only needing to specify the fields that I care about, and was hoping it would be able to generate very compact machine code (i.e. a single store to the 32 bit register). However, the generated code was this:

    ldr r3, [sp, #4]
    ldr r2, [pc, #76]
    bfc r3, #0, #1
    str r3, [sp, #4]
    ldr r3, [sp, #4]
    bfc r3, #1, #1
    str r3, [sp, #4]
    ldr r3, [sp, #4]
    bfc r3, #2, #1
    str r3, [sp, #4]
    ldr r3, [sp, #4]
    bfc r3, #3, #1
    str r3, [sp, #4]
    ldr r3, [sp, #4]
    bfc r3, #4, #1
    str r3, [sp, #4]
    ldr r3, [sp, #4]
    bfc r3, #5, #2
    str r3, [sp, #4]
    ldr r3, [sp, #4]
    bfc r3, #7, #1
    str r3, [sp, #4]
    ldr r3, [sp, #4]
    bfc r3, #8, #2
    str r3, [sp, #4]
    ldr r3, [sp, #4]
    bfc r3, #10, #22
    str r3, [sp, #4]
    ldr r3, [sp, #4]
    str r3, [r2, #0]

It appears that because in the svd2ada generated interface the register type itself is declared Volatile_Full_Access, in constructing the record value to assign to the register, the compiler repeatedly loads and stores the intermediate value to the stack before finally assigning it to the register.

I was wondering if there was a reason that the register type was declared Volatile_Full_Access, instead of the fields in the peripheral types. It seems to me that declaring it on the peripheral fields instead of the register types would more accurately capture the intent of adding Volatile_Full_Access. Doing so allows the compiler to generate the following instead:

    movs r3, #0
    bfc r3, #0, #1
    bfc r3, #1, #1
    bfc r3, #2, #1
    bfc r3, #3, #1
    bfc r3, #4, #1
    bfc r3, #5, #2
    bfc r3, #7, #1
    bfc r3, #8, #2
    ldr r2, [pc, #8]
    bfc r3, #10, #22
    str r3, [r2, #0]

This is definitely better because the useless loads and stores are gone, though still very wasteful. At this point however, this seems to be more of poor optimization on the compiler's part than any fault of svd2ada. It seems that the compiler (which perhaps I should mention is the bare board arm compiler from GPL2017) is unable to fold the operations for setting the different bits when the fields don't align to byte boundaries (I ran another test with my own type where things did align to byte boundaries and the bfcs disappeared).

In the end, after a fair amount of experimentation, the most reliable way I've found to ensure that the optimal code is generated was to do an Unchecked_Conversion from a UInt32 to the register type, but this kind of defeats the point of having the interface provided by svd2ada.

Still, it seems to me like modifying the code generating to apply Volatile_Full_Access to where the registers are defined in the peripheral records, as opposed to applying it to the register type as a whole could be a worthwhile improvement, as it could allow people to more efficiently work with temporary copies of registers.

LordPython, 2018-02-11:
I just tested the same code, but compiling with gcc 7, and for whatever reason, it did a much better job of optimization, and came up with this assembly

    movs r2, #0
    ldr r3, [pc, #4]
    str r2, [r3, #0]

Which is what I was hoping for in the first place. But it does seem that to generate this it is necessary to move the Volatile_Full_Access aspect.

simonjwright, 2018-02-12:
I tried this, and moving the aspect to the register field in the device worked as @LordPython suggests. However, it worked for me with all of GNAT GPL 2017, GCC 7.1.0, and GCC 8.0.1. See attached.
[sizing.zip](https://github.com/AdaCore/svd2ada/files/1716608/sizing.zip)

lambourg, 2018-02-14:
This is indeed a very good remark, and a flaw in the current schema. The Volatile_Full_Aspect aspect should only be applied to the peripheral fields, so that aggregates can be generated efficiently (and any local copy of a register can also benefit from compiler optimisations)

I'll fix that and will let you know. Thanks for the report!

lambourg, 2018-09-17:
Hi @LordPython,

gcc-7 indeed fixed some issues there, in particular when built with -O2 (of course, at -O0 the generated code will be bloated).

So I'm closing this issue, in particular now that the GNAT community release is done.

Best regards,
- Jerome

closed 2018-09-17 by lambourg