AdaCore/svd2ada #47: Enum conflict between the Enum name and the Enum member

corentingay, 2018-04-02:
When an Enum is dumped, the code checks that there is no dupplicates with the same name in the Ada spec, however, if one of the name of its children is '1' (or any number for that matter), the new name of the enum will conflict with the name of its member causing a compilation error.

For an example of that. see the following code :

```ada
type CTRLA_FORMSelect_1 is
     (
      --  USART frame
      CTRLA_FORMSelect_0,
      --  USART frame with parity
      CTRLA_FORMSelect_1)
```

The CMSIS-Pack used is "Atmel SAM D20 Series Device Support and Examples"
found on that page "https://www.keil.com/dd2/pack/".

The file passed to svd2ada is 'SVD/SAMD20/ATSAMD20J18.svd'
I have a pull request but it needs more testing, it should be here in a few days.

closed 2018-04-30 by lambourg

Joebeazelman, 2022-02-27:
I got this error and had to remove the name from the enum and then create a function that returns it as an alias to the its with the duplicate value. 

https://stackoverflow.com/questions/18486864/ada-enumerations-with-duplicate-values

Fabien-Chouteau, 2022-02-28:
@Joebeazelman  this looks like a different problem.

Please open a separate issue with more details like input SVD and output code.

Thanks,