AdaCore/svd2ada #61: Bit order always low

simonjwright, 2019-09-05:
At descriptors-register.adb:315 it says
```
Add_Bit_Order_Aspect (Rec, System.Low_Order_First);
```
(this is the only use of `Add_Bit_Order_Aspect`).
I dare say there hasn’t been any need to work with big-endian processors?

pat-rogers, 2019-09-05:
Should be OK. Ada 2012 is different from 2005 in this regard, in that both orders are allowed for either endianess, as long as a couple of requirements are met to prevent overlapping byte boundaries (RM 13.5.1/10).

simonjwright, 2019-09-06:
Isn’t it possible for a peripheral to be big-endian? If not, and if we’re only dealing with little-endian MCUs, why do we need to mention the endianness at all?

pat-rogers, 2019-09-09:
On 06-Sep-19 11:47 AM, Simon Wright wrote:
> Isn’t it possible for a peripheral to be big-endian? If not, and if 
> we’re only dealing with little-endian MCUs, why do we need to mention 
> the endianness at all?

The bit order specification only indicates what the programmer means by 
the bit numbers.

pat-rogers, 2020-06-02:
Simon, any other comments here?

simonjwright, 2020-06-02:
No, if you’re all happy with the present situation, that’s OK

closed 2020-06-02 by pat-rogers