AdaCore/svd2ada #63: Volatile Full Access suppresses use of arrays

simonjwright, 2020-03-13:
`--no-vfa-on-types` effectively forces `--no-arrays`, presumably because of the [GNAT RM](https://gcc.gnu.org/onlinedocs/gnat_rm/Pragma-Volatile_005fFull_005fAccess.html#Pragma-Volatile_005fFull_005fAccess)

>It is not permissible to specify Volatile_Full_Access for a composite (record or array) type or object that has an Aliased subcomponent.

This is a shame! would be nice to have `Volatile_Full_Access_Components` on the lines of `Volatile_Components` ([ARM K.1(68)](http://www.ada-auth.org/standards/rm12_w_tc1/html/RM-K-1.html#p68)).

lambourg, 2020-03-13:
Hello Simon,

Things will improve with Ada2020, in particular [AI12-0128-1](http://www.ada-auth.org/cgi-bin/cvsweb.cgi/ai12s/ai12-0128-1.txt). Basically Volatile_Full_Access will become OBE and will be replaced by a more precise definition of Atomic.

simonjwright, 2020-03-13:
OK, of course.

Would it be possible to amend the `svd2ada --help` for `--no-vfa-on-types` to say that `--no-arrays` is forced? (and, personally, I’d like to see a 79-column limit on the help output :-)

pat-rogers, 2020-03-15:
On 13-Mar-20 10:15 AM, Simon Wright wrote:
> OK, of course.
> 
> Would it be possible to amend the |svd2ada --help| for 
> |--no-vfa-on-types| to say that |--no-arrays| is forced? (and, 
> personally, I’d like to see a 79-column limit on the help output :-)

Seems reasonable indeed.

pat-rogers, 2020-06-01:
The forcing of --no-arrays is now indicated in the help message.  
Those messages have been shortened overall.

closed 2020-06-02 by pat-rogers