AdaCore/svd2ada #65: Why bit field value name casing is modified ?

NicoPy, 2020-05-10:
Here is an example of the description of a bit field generated by svd2ada :

```ada
   --  VREF Clock Gate Control
   type SCGC40_VREF_Field is
     (--  Clock disabled
      Clockdisabled,
      --  Clock enabled
      Clockenabled)
     with Size => 1;
   for SCGC40_VREF_Field use
     (Clockdisabled => 0,
      Clockenabled => 1);
```
`Clockdisabled` and `Clockenabled` are defined respectively as `ClockDisabled` and `ClockEnabled` in the svd file.
Why is the casing modified ? 
This makes field names hard to read. Especially long ones.
Try to read this one Flashisdisabledduringwaitmode or this one K4xfamilyork2xfamilywithtamper ;)

Fabien-Chouteau, 2020-05-11:
Hello @NicoPy,

> Why is the casing modified ?

`svd2ada` generates code that follow the GNAT coding style so the identifiers use combination of snake and camel case: `Like_This`

> This makes field names hard to read. Especially long ones.
> Try to read this one Flashisdisabledduringwaitmode or this one K4xfamilyork2xfamilywithtamper ;)

We could either convert everything to snake camel case by inserting underscores before capital letters: `Clock_Enabled`

Or have the option to keep casing: `ClockEnable`

CC @pat-rogers @lambourg

NicoPy, 2020-05-11:
Why not add an option to select which style to use during conversion ?

Fabien-Chouteau, 2020-05-11:
Yes that is a possibility.

pat-rogers, 2020-05-11:
>     This makes field names hard to read. Especially long ones.
>     Try to read this one Flashisdisabledduringwaitmode or this one
>     K4xfamilyork2xfamilywithtamper ;)
> 
> We could either convert everything to snake camel case by inserting 
> underscores before capital letters: |Clock_Enabled|

Although that is the most readable, IMO, does it not open the 
possibility of introducing homonyms?  Or can we know that both styles 
are never used together in the same input file? In this case perhaps 
not, although I have seen it done in user-written code.

> Or have the option to keep casing: |ClockEnable|

That seems like the right default to me, under the circumstances. Not 
that I find that style very readable either.

Matthias-R, 2020-05-11:
> That seems like the right default to me, under the circumstances. 

Definitely yes. The reference when working with a microcontroller is the datasheet and/or the programming manual - and the naming in the .svd file is [should be] identical to the manual.
It would be a nightmare when the .ads file was different - you have to guess where underscores have been inserted.
I don't really like CamelCase either, but in my opinion it is more readable than everythinginlowercase.

pat-rogers, 2020-05-12:
On 11-May-20 12:58 PM, Matthias Richter wrote:
> That seems like the right default to me, under the circumstances.
>
> Definitely yes. The reference when working with a microcontroller is the
> datasheet and/or the programming manual - and the naming [...]

Agreed.

> I don't really like CamelCase either, but in my opinion it is more 
> readable than everythinginlowercase.

My favorite is spaceperson, which could be either space_person or 
space_per_son. :-)

NicoPy, 2020-05-24:
So, what's the final decision ?

pat-rogers, 2020-05-27:
On 24-May-20 8:09 AM, Nicolas Pinault wrote:
> So, what's the final decision ?

We'll make it retain the casing encountered in the SVD input.

NicoPy, 2020-05-28:
Ok. Great.
Thanks for your support.

pat-rogers, 2020-05-28:
Our pleasure.

pat-rogers, 2020-06-02:
The letter casing for enumerals is now retained unchanged from the SVD input file.

closed 2020-06-02 by pat-rogers

pat-rogers, 2020-06-02:
On 28-May-20 11:23 AM, Nicolas Pinault wrote:
> Ok. Great.
> Thanks for your support.

You're very welcome.

Note that the change has been applied to enumeration type values, 
specifically.  I didn't see any other occurrences but may have missed 
one. Let me know if you see any others.

NicoPy, 2020-06-03:
I have not noticed any other occurrence till now. I'll let you know if I find one.
Thanks again.