AdaCore/svd2ada #70: Register name "range"

NicoPy, 2020-06-12:
When the svd file to be converted contains a register named `range`, svd2ada does not generate any error or warning.
Of course, the resulting Ada file does not compile.
Does svd2ada check for peripheral/register/field names to not use reserved Ada keywords ?

Fabien-Chouteau, 2020-06-12:
Hi @NicoPy,

It should be handled here: https://github.com/AdaCore/svd2ada/blob/b5aa8299df974993993b9c767ecb171a7bc4a53d/src/ada_gen.adb#L107

NicoPy, 2020-06-12:
Bonjour  @Fabien-Chouteau ,
Hmmm, looking at the code you pointed me to, I now understand my error.
The name is effectively modified so it does not conflict with reserved keywords.
I believe it would be a good idea to have an option to show warnings about modified identifiers.

I will not tell you why I made a bad diagnosis (maybe another story with a cat walking on a keyboard) ;)
Sorry for the inconvenience.

closed 2020-06-12 by NicoPy