AdaCore/svd2ada #80: Bad content in handler.S when converting a SVD file for CortexM7.

NicoPy, 2020-12-17:
Hi,
When converting a SVD file for a CortexM7 µP, the generated content of handler .S is not correct :
```
	.syntax unified
	.cpu cortex-m4
	.thumb
```
Any fix or workaround ?

Regards,
Nicolas

Fabien-Chouteau, 2020-12-18:
Hello @NicoPy ,

Why do you consider it not correct? cortex-m7 is backward compatible with cortex-m4, specially for such simply ASM code.

NicoPy, 2020-12-18:
Hi @Fabien-Chouteau ,
Sure, cortex-m7 is a superset of cortex-m4, but...
- Will the compiler be happy when you tell it you compile for cortex-m7 while the asm file tells it is made for cortex-m4 ?
- This is not Ada spirit. A signal space can make the compiler complain. Why should we accept such a non coherency ?

Fabien-Chouteau, 2021-01-04:
> * Will the compiler be happy when you tell it you compile for cortex-m7 while the asm file tells it is made for cortex-m4 ?

Yes.

> * This is not Ada spirit. A signal space can make the compiler complain. Why should we accept such a non coherency ?

As explained above the there is no incoherency here.

You can contribute to this pull-request if you want to work on different behavior for the tool: https://github.com/AdaCore/svd2ada/pull/50
The main problem being that SVD files do not document the CPU core.

closed 2021-01-04 by Fabien-Chouteau