AdaCore/svd2ada #89: Comment fields should come after the generated code and not before

kevlar700, 2022-01-18:
If you look at the following files in an IDE and especially line 741 of the first link (stm32-device.adb).
         `RCC_Periph.APB2RSTR.SDMMC1RST := True;`

"https://github.com/AdaCore/Ada_Drivers_Library/blob/master/arch/ARM/STM32/devices/stm32f7x9/stm32-device.adb"
"https://github.com/AdaCore/Ada_Drivers_Library/blob/master/arch/ARM/STM32/svd/stm32f7x9/stm32_svd-rcc.ads"

And hover/highlight over e.g. `SDMMC1RST` in an IDE then you will see the wrong comment (SPI 1 reset) instead of "SDMMC2 reset".

Generally not much of an issue but it could be confusing when the names are less obvious.

kevlar700, 2022-03-15:
Record comments are best left where they are as they do not seem to be interpreted by IDEs. I'm not sure about other types but this commit solves most if not all of the practical issue for me by changing just the comment location of record fields. Variant record comments stay in the same place above. They should also move below but I haven't found where they are generated, yet.

https://github.com/kevlar700/svd2ada/commit/255c5786ec34cbd21a202852138ce1a28573f312

kevlar700, 2022-03-15:
Found the discrimnated variant record fix
https://github.com/kevlar700/svd2ada/commit/255c5786ec34cbd21a202852138ce1a28573f312

pat-rogers, 2022-03-16:
Really I think this will be tricky because it is up to the IDE to decide 
which line(s) to display. IIRC, the choice has changed over time.

I wonder if there is an IDE option to control it.

kevlar700, 2022-03-17:
Interesting. I shall look out for IDE options but I believe that was my first port of call. I suppose the other option is a runtime flag. These patches make the record comments correct for me on GNAT Studio and VsCode and so may be helpful to others, atleast.

kevlar700, 2024-03-04:
I have the same issue with enumerated values now which I did not know svd2ada supported until recently. ST have started putting much longer descriptions in their SVD files, which is nice too.
I guess any kind of consensus was not forthcoming?

kevlar700, 2024-03-04:
The leading setting may well be the answer. It is easy to apply to vscode but I am having difficulty with Gnat Studio.

https://github.com/AdaCore/ada_language_server/blob/master/doc/settings.md#documentationstyle

kevlar700, 2024-07-02:
This seems to be working now in the latest Gnat Studio CR 25.0w (20240506) simply by setting the check box for leading documentation in preferences => Documentation and restarting Gnat Studio or the language server. I assume that I tried restarting Gnat Studio before after setting it but in any case I am quite happy with using this as my default now. I shall close this issue but feel free to re-open it if you still think there is an issue to deal with in terms of svd2adas default.

closed 2024-07-02 by kevlar700