AdaCore/svd2ada #90: No script or documentation for integrating it into the Ada toolchain

Joebeazelman, 2022-02-24:
It would be nice if the build scripts collected the build artifacts into a single application folder containing all the necessary dependencies which can be installed as part of the build chain.

Fabien-Chouteau, 2022-02-25:
Which build artifacts are you referring to?

Joebeazelman, 2022-02-25:
After downloading the repository and building it, there's no instruction or scripts for deploying it into the system. I presume just copying the svd2ada executable to my local bin isn't enough. It probably needs additional directories and files. Ideally, I wish I could install it using ALR, but I couldn't find it as a package.

Fabien-Chouteau, 2022-03-01:
Hello @Joebeazelman 

svd2ada is not available in Alire!

closed 2022-03-01 by Fabien-Chouteau