AdaCore/svd2ada #92: SVD2ADA generates enumeration with duplicate values

Joebeazelman, 2022-02-28:
After generating and compiling the code generated from the enclosed SVD file, the Ada compiler flagged an error because two of the names in an enumeration had the same value. I worked around the error by deleting the name from the enum and returning it from a function as an alias to the other name with the same value.

[LPC176x - 5x SVD.zip](https://github.com/AdaCore/svd2ada/files/8155885/LPC176x.-.5x.SVD.zip)

pat-rogers, 2022-02-28:
On 28-Feb-22 1:09 PM, Joebeazelman wrote:
> After generating and compiling the code generated from the enclosed SVD
> file, the Ada compiler flagged an error because two of the names in an
> enumeration had the same value. I worked around the [...]

Sounds like a bug in the SVD file indeed.

Joebeazelman, 2022-03-01:
Enum names can have the same value in other languages, but not in Ada. It would be a bug if the SVD specification forbids it.