AdaCore/svd2ada #94: I get a series of errors when trying to compile svd2ada

MilliUser, 2022-09-17:
Hello. I'm so new to ADA I'm a novice at some places. I found answers to some of my problems on the internet, but I couldn't get past the installation problem in svd2ada. I don't know if it is due to the file directory or some other problem.

The errors I get are after " gprbuild -P svd2ada.gpr ".
PS C:\GNAT\svd2ada-master> gprbuild -P svd2ada.gpr
Compile
[Ada] svd2ada.adb
[Ada] ada_gen.adb
[Island] descriptors.ads
[Ada] descriptors-device.adb
[Ada] svd2ada_utils.adb
[Ada] base_types.adb
[Ada] base_types-register_properties.adb
[Island] descriptors-peripheral.adb
[Ada] descriptors-cluster.adb
[Ada] descriptors-register.adb
[Island] descriptors-field.adb
[Ada] descriptors-enumerate.adb
bind
[gprbind] svd2ada.bexch
[Ada] svd2ada.ali
error: "s-imglld.ali" not found, "s-imglld.adb" must be compiled
error: "s-valenu.ali" not found, "s-valenu.adb" must be compiled
error: "dom.ads" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "dom-core.adb" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "sax.ads" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "sax-encodings.ads" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "unicode.adb" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "unicode-names.ads" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "unicode-names-basic_latin.ads" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "unicode-ces.adb" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12
error: "unicode-ces-utf32.adb" and "svd2ada.adb" compiled with different GNAT versions, v10 and v12.....
it goes on and on...

simonjwright, 2022-09-18:
Well, it looks as though you have two different GNAT versions on your Windows computer :-)
I’d guess that you’ve got GNAT CE 2021 (that would be the v10) and Alire (that would be the v12).
Most of the files you have problems with are part of XML/Ada, and with GNAT CE 2021 those are supplied pre-built with that compiler. With Alire, they’re fetched as source and compiled with the selected Alire toolchain..

You say `gprbuild -P svd2ada.gpr`, which will use the GNAT CE 2021 `gprbuild`, which knows to look for XML/Ada as part of the GNAT CE 2021 installation. Instead, say `alr build`.
_I must say I don’t know how `gprbuild` finds the Alire-installed GNAT, I’d have expected it to use the GNAT CE 2021 compiler!_

MilliUser, 2022-09-18:
I wanted to develop with VS Code. I followed the installation instructions in Using Ada in VS Code.
I installed gnat compiler with MSYS. Do I need to uninstall this installation?

simonjwright, 2022-09-18:
Sorry, I don’t know VS Code or MSYS.

MilliUser, 2022-09-18:
thank you. I guess I have to start all over again

closed 2023-01-02 by MilliUser