AdaCore/svd2ada #96: Discriminated type cannot be volatile in SPARK mode

kevlar700, 2022-11-14:
After discovering that Spark supports overlays. I was just investigating whether it would be possible for svd2ada generated files to pass SPARK mode compliance checks.

With arrays off. The only error I see atleast before compilation fails is discriminated type cannot be volatile.

Is there a compatible alternative generative option that could be added to svd2ada command flags? Or is it better to just exclude these files. Perhaps svd2ada could generate packages with Spark mode off instead?

kevlar700, 2022-11-15:
After experimentation it seems that Spark mode off on the generated files seems to cause more Spark mode violations and not less. Obviously re-open it if you see a way forward.

kevlar700, 2023-11-29:
I had a look at optionally removing discriminants from arrays but it requires more changes and understanding than I can currently afford the time for when I can just use unchecked_conversions for the few long arrays.

The relevant code for creating arrays is at
descriptors-field.adb line 496 (Union_T)
ada_gen.adb line 2212 (New_Type_Union)

simonjwright, 2023-11-29:
On 29 Nov 2023, at 14:27, Kevin Chadwick ***@***.***> wrote:
> 
> I have seen @simonjwright <https://github.com/simonjwright> use --no-arrays. I am not sure how he avoids e.g. 30 item case statements for EXTI/Interrupts etc.. Unless he overlays a 32 bit value with an address.

Where was that?

kevlar700, 2023-11-30:
> Where was that?

I am not sure. My memory is usually very good but I can be mistaken. I guess it was in a particular context then, if at all.

kevlar700, 2024-03-04:
Update: I decided to simply create multiple peripherals e.g. TIM1_Output_Peripheral,TIM1_Output_Periph and TIM1_Input_Peripheral,TIM1_Input_Periph and manage the patching manually for now. Even on the new U5 chip that we are using there are only 7 files to manage instead of 2 with L4.