AdaCore/svd2ada #98: svd2ada crashes if `derivedFrom` references tags defined after current tag and always if used for enumeration values

luzidchris, 2022-12-12:
I tried to generate code from this SVD:
https://raw.githubusercontent.com/esp-rs/esp32/master/svd/esp32.svd

`svd2ada` even crashes if the tag which is referenced by `derivedFrom` is not listed above the tag which make use of `derivedFrom`. It's not really nice to fix this manually but it works but I got stuck because `derivedFrom` is also used in an enumeration value - although the field the tag references is located before. `svd2ada` crashses:

```txt
raised CONSTRAINT_ERROR : enumerate 'derivedFrom' is not known: WDT_STG0
[.../svd2ada]
0x4d1443 descriptors__enumerate__read_enumerate at ???
0x4e3e84 descriptors__field__read_field at ???
0x4d6c6d descriptors__register__read_register at ???
0x5197d0 descriptors__peripheral__read_peripheral at ???
0x521b4f descriptors__device__read_device at ???
0x451f1b _ada_svd2ada at ???
0x450e34 main at ???
[/lib/x86_64-linux-gnu/libc.so.6]
0x7fc289c03d8e
0x7fc289c03e3e
[..../svd2ada]
0x450e8c _start at ???
0xfffffffffffffffe
```
Besides of `-p` and `-o`, I've not passed any further arguments to `svd2ada`.

I'd really be happy if there was solution to fix or work-around this problem... 

I've found issue #84 but it seems different to me ... (though there's not been any reaction on this issue yet).

luzidchris, 2022-12-12:
Further info:

First, `svd2ada` complained about `I2C`, `TIMG` and `MCPWM`. I could resolve this be moving tags as mentioned before.

```
raised CONSTRAINT_ERROR : peripheral 'derivedFrom' is not known: I2C
```

```
raised CONSTRAINT_ERROR : peripheral 'derivedFrom' is not known: TIMG
```

```
raised CONSTRAINT_ERROR : peripheral 'derivedFrom' is not known: MCPWM
```

Eventually, I got stuck with this:

```
raised CONSTRAINT_ERROR : enumerate 'derivedFrom' is not known: WDT_STG0
```

`WDT_STD0` is an enumeration:

```xml
                        <field>
                            <name>WDT_STG0</name>
                            <bitOffset>29</bitOffset>
                            <bitWidth>2</bitWidth>
                            <enumeratedValues>
                                <name>WDT_STG0</name>
                                <usage>read-write</usage>
                                <enumeratedValue>
                                    <name>Disable</name>
                                    <description>Disabled</description>
                                    <value>0</value>
                                </enumeratedValue>
                                <enumeratedValue>
                                    <name>Interrupt</name>
                                    <description>Trigger an interrupt</description>
                                    <value>1</value>
                                </enumeratedValue>
                                <enumeratedValue>
                                    <name>ResetCPU</name>
                                    <description>Reset CPU core</description>
                                    <value>2</value>
                                </enumeratedValue>
                                <enumeratedValue>
                                    <name>ResetSystem</name>
                                    <description>Reset System</description>
                                    <value>3</value>
                                </enumeratedValue>
                            </enumeratedValues>
                        </field>
```

And other enumerations need to reference it:

```xml
                        <field>
                            <name>WDT_STG1</name>
                            <bitOffset>27</bitOffset>
                            <bitWidth>2</bitWidth>
                            <enumeratedValues derivedFrom="WDT_STG0" />
                        </field>
```

luzidchris, 2022-12-13:
I've taken a look at the implementation of 'svd2ada' and did some tests. It seems to me that the issue is caused by `Read_Enumerate`. The functions is passed a vector of type `Enumerate_Values_Vectors.Vector`. Package `Descriptors.Fields` calls the function in context of `Read_Field` like this (for tag `enumeratedValues`):

```ada
                     Enum : constant Descriptors.Enumerate.Enumerate_T :=
                              Descriptors.Enumerate.Read_Enumerate
                                (Child, Result.Enums, Result.Acc = Write_Only);
```
Vector `Result.Enums` is used by `Read_Enumerate` to check for an existing type it can derive from if `derivedFrom` is not empty. In my opinion this is the wrong vector because `Result.Enums` contains the enumeration value tags, not the `enumerationValues` tags which is the parent of one or more enumeration value tags.
To find the `enumerationValues` tag another tag references using `derivedFrom`, it would be necessary to search `Vec` which is passed to `Read_Field`:

```ada
   function Read_Field
     (Elt            : DOM.Core.Element;
      Vec            : Field_Vectors.Vector;
      Default_Access : Access_Type;
      Default_Read   : Read_Action_Type)
      return Field_T;
```
I ran out of time to further analyze the issue but I'll come back to this as soon as I know more. I just wanted to quickly share my view on this before anyone spends effort on this..

luzidchris, 2022-12-27:
Just spend some more time on this but it's not possible to quickly fix this because it's necessary to search the list of enumeration values of all registers. As mentioned before, `Read_Enumerate` is only passed the list of enumerations of the register the current element belongs to. In my case, the enumeration value which serves as template of the current one is defined by one of the previous registers and the enumeration values of this one are not visible to `Read_Enumerate`.

I guess some sort of callback is necessary to query the upper layer(s) for the type an item is derived from but I'm afraid that would require some rework to avoid circular dependencies. The variable `Vec` I mentioned in the previous post is not enough since it only contains the enums of the current field. To obtain an enum of other fields, it would be necessary to traverse all registers, or even all peripheral tags. At least if that's the purpose of `derivedFrom`.

kevlar700, 2024-10-03:
Not a fix exactly but I have found that you can add a matching <name> within two <enumeratedValues> in order to provide one generated enum type to two registers with the same values.

RREE, 2026-08-21:
I also see this error when running svd2ada for the esp32. As far as Claude tells me, the ESP32.svd is valid CMSIS (a named enumeratedValues can be shared across fields in the same register/peripheral).

Here Claude's more detailed analysis:
The bug is in src/descriptors-field.adb:129-135:

```Ada
elsif Tag = "enumeratedValues" then
   declare
      Enum : constant Descriptors.Enumerate.Enumerate_T :=
               Descriptors.Enumerate.Read_Enumerate
                 (Child, Result.Enums, Result.Acc = Write_Only);
```
When parsing WDT_STG2's enumeratedValues, it passes Result.Enums — the current field's own, still-empty, enum vector — into Read_Enumerate. That function (src/descriptors-enumerate.adb:141-168) only searches whatever vector it's handed for a name match; since WDT_STG2's own Result.Enums is empty at this point, the lookup for "WDT_STG3" fails and it raises Constraint_Error.

Note that Read_Field does have access to Vec (the register's already-parsed sibling fields, each carrying its own .Enums), and correctly uses it for field-level derivedFrom="..." on <field> (register.adb passes Result.Fields in as Vec, descriptors-field.adb:65-73). But that sibling-field vector is never consulted for the nested enumeratedValues derivedFrom, only the field's own local enum list is — which is the gap.

I will see if Claude also helps to fix this