AdaCore/svd2ada #99: `range check failed` because type `Natural` is used for address values

luzidchris, 2022-12-16:
Some of the SVD files in `CMSIS_SVD` lead to `CONSTRAINT_ERROR` exceptions because type `Natural` is too small to store address values. 
This can be reproduced for example with [M1AGL.svd](https://github.com/AdaCore/svd2ada/blob/master/CMSIS-SVD/Microsemi/M1AGL.svd).

The following error is raised by `svd2ada`: 

```
raised CONSTRAINT_ERROR : base_types.adb:218 range check failed
```
In my system (linux 64-bit), type `Natural` has a max value of 2147483647, i.e. 0x7FFFFFFF. As a result of this, `svd2ada` fails to process address or size values which are greater than this upper limit.

Fabien-Chouteau, 2023-01-02:
Thanks @luzidchris, do you know if SVD defines limits for the types of those values?

luzidchris, 2023-01-06:
Hi, 

I've unfortunately not found any constraints. I assume this is not defined by SVD format because memory mapping is pretty target-specific.

In my case, the `Constraint_Error` was raised while calculating an absolute address and therefore my initial idea was to align data types to `Unsigned` (defined by Base_Types if I remember correctly). The type is already used for some of the address-related calculations, it's in `svd2ada`. And it should ensure that calculations work for all (or at least 99,9999 %) of the existing target architectures (incl. 64-bit).

Fabien-Chouteau, 2023-01-09:
Maybe we should switch to the Ada2022 Big_Numbers for those values.