AdaCore/svd2ada #110: ADA.IO_EXCEPTIONS.NAME_ERROR when executing Svd2Ada

Joebeazelman, 2025-01-21:
I installed Svd2Ada using Alire's install command:

Installation prefix found at /Users/Joe/.alire
Contents:
   svd2ada=0.1.0
   xmlada=22.0.0
   xtensa_dynconfig=0.1.0-dev

To use the program, I exported Alire's installation location.  When I execute svd2ada on a SVD file, I get the following error:

```
svd2ada --package=esp32s3.svd --gen-interrupts --no-uint-subtypes ../svd/esp32s3.svd --output=../esp32s3_svd

Execution of svd2ada terminated by unhandled exception
raised ADA.IO_EXCEPTIONS.NAME_ERROR : Could not open /Users/Joe/.alire/schema/CMSIS-SVD_Schema_1_3_1.xsd
Load address: 0x104cbc000
Call stack traceback locations:
0x104e6e31c 0x104e6e37c 0x104e5dbb0 0x104cbf4d0 0x104cc034c 0x104ec173c
```

The file doesn't exist

simonjwright, 2025-01-22:
This could almost certainly be fixed if PR #108 was installed (only 4 months old)

ThyMYthOS, 2026-10-06:
Please consider making a 0.2.0 release and put it into Alire index.